Snapmaker/Luban #249: G4 is not interruptible.

tonejaco, 2020-03-18:
I found that the G4 was not interruptible with the "Stop" button of the Touch Display. The only thing I could do was turn off the power supply. I think the Stop function should be able to interrupt anything and everything, similar to having an emergency stop function.

tonejaco, 2020-03-23:
Other operations are not uninterruptible too like waiting for the bed or nozzle to heat up.

parachvte, 2020-04-09:
We are refactoring firmware code, by separating Marlin task and HMI task we will be able to interrupt G-code easily.

tonejaco, 2020-04-09:
I was thinking that some of the problem lies with the queue in Marlin. I don't know much about it but I was thinking they almost need two queues, one for high priority tasks and one for informational and other type low priority tasks that can be executed any time it is not busy.

parachvte, 2020-04-09:
Exactly! We did have another queue for commands from Touchscreen, but it's not enough to solve the the problems we faced. We are trying to migrating FreeRTOS into Marlin.

tonejaco, 2020-04-09:
Super. Just what is needed. Is the Marlin community interested in that too? Seems like they should be.

parachvte, 2020-04-10:
There were discussions on Marlin over RTOS, I quoted the author's comment here:

> RTOS doesn't appear to add any value to this application with its AVR-centric code architecture. A complete bottom-up rewrite that uses an RTOS would be possible, and that would be the proper approach. But then it would be a completely new firmware.

If there will be a Marlin 3.0, probably they will rebuild on RTOS. Who knows.

tonejaco, 2020-06-24:
Is this now possible now that RTOS is there?

parachvte, 2020-07-02:
From firmware 1.9.0, FreeRTOS is there.

tonejaco, 2020-07-02:
Are you saying that with 1.9.0 the G4 command can now be interrupted? (i.e. "Stopped")?

parachvte, 2020-07-03:
Hmmmm no. i'm saying we can now facing the staff like Wi-Fi echo issue and interruptions.

parachvte, 2020-12-01:
Moved to the controller repo.

closed 2020-12-01 by parachvte