Snapmaker/Luban #250: Full G & M code listing in console

tonejaco, 2020-03-18:
List the FULL set of G & M codes available in Luban console. I understand that many of these are for expert or debugging use and they should be highlighted as such.

tonejaco, 2020-03-30:
Maybe have a switch to list the expert features highlighted differently.

parachvte, 2020-04-07:
I'm tweaking on https://github.com/snapmaker/Documentation to document a full G-code reference for Snapmaker version of Marlin.

Note that this is only an initial setup, I'll add/remove G-code to align with the source code.

tonejaco, 2020-04-07:
What is the planned location for this documentation to exist?

parachvte, 2020-04-08:
I planned to leave it as a Github Page, and also mounted as child node on https://snapmaker.com/document

tonejaco, 2020-04-08:
You perhaps already know this but there are a bunch of links in the current set of docs that refer to the "/docs" directory that need to be changed to "/Documentation ",

parachvte, 2020-04-09:
The "/docs" links are actually link to original Marlin document, considering that cost of adding these docs is a bit large, I would leave them there...

tonejaco, 2020-04-09:
Can you just rename the directory to match?

parachvte, 2020-04-09:
I re-checked the links.
For the G-code links, I definitely will correct their links.
For other link like configurations (MACRO definition, etc) and explanations, I'll ignore them.

tonejaco, 2020-04-09:
Is there a single file that defines what is in the G/M code list?