Snapmaker/Luban #251: Console not echoing thru WiFi connection

tonejaco, 2020-03-18:
I understand that this is a known issue, just want to track its repair.
When executing commands thru the console, the results are not echoed back when connected thru WiFi.

tonejaco, 2020-04-02:
What's the plan on this issue?

parachvte, 2020-04-09:
This issue is in firmware, also planned after our refactor.

tonejaco, 2020-04-11:
I just noticed that while connected via WiFi and hitting the "home" button in the "Control" window in Luban echos back the following:
G53
ok
G28
ok
G54
ok

Is that just Luban doing that or is the controller actually echoing what it is suppose to???
If that one works, why don't the others? Maybe just a clue.

parachvte, 2020-04-12:
The behavior is done by Luban, we switch back to use G54 coordinate system after Homing.

Calling G28 will reset offsets regarding all 3 axes, so a G54 to reload offset is needed.
This is the default behavior of in recent Luban updates.

tonejaco, 2020-04-13:
Hmm. So I have the 3DP on, why am I seeing G54?

tonejaco, 2020-05-09:
WHY IS THIS STILL NOT FIXED!!!
I'm disappointed! Please pass this on to the Firmware Team.:frown:

Also, the console echoes back anything you type with an "ok" (FW 1.8)

scotthsl, 2020-06-23:
[image]

Tone, attached is the communication link for Snapmaker2. As you see, all Marlin original output will send to PC1 over UART1. And if you communicate with controller thru WiFi, Gcodes will be wrapped in SM2 protocol, and then send to main controller. Original Gcode output will be send out over UART1. Main contoller just send a ACK with SM2 protocol format to screen, then screen reply to Luban.

scotthsl, 2020-06-23:
So you can't get original Gcode output thru WiFi. We plan to implement it in next or next two release. Because this function may have more changes.

parachvte, 2020-12-01:
Moved to the controller repo.

closed 2020-12-01 by parachvte