Snapmaker/Luban #253: Feature Request: Simple text editor as a new Tab in Luban

tonejaco, 2020-03-18:
A new tab that is a simple text editor would be a great addition to the Snapmaker Luban App. Then the user doesn't have to go to some external editor which may introduce bad text formatting. Future versions could do syntax checking and have shortcuts to inserting G & M code commands. All the other tabs could have an option to open the gcode file in the text editor.

tonejaco, 2020-03-20:
Or maybe a Widget in the workspace?

parachvte, 2020-03-22:
Text editor could be too complicated if you want syntax support.

Is there any existing editor for G-code?

Ralnoc, 2020-03-22:
I haven't been able to find an editor, though prismjs is an extendable syntax highlighter that could possibly be adapted to be integrated into a custom editor.

tonejaco, 2020-03-22:
It’s just plain text. Start basic. Add features later. If it’s a large file then it will need to be paged in.

tonejaco, 2020-03-29:
The Macro Widget is kind of the thing, only it can't open, close & save to/from files.
It's also doing it in a pop up window instead of it's own pane.