Snapmaker/Luban #254: Feature Request: Copy , Cut & Paste in Console

tonejaco, 2020-03-18:
This is likely on your list already, mentioned by a lot of people. Just want to track it.

tonejaco, 2020-04-02:
How hard would it be to just start with being able to copy the text in the console window?
I get so tired of having to transcribe the matrix data. Especially with a 5 x 5 or larger matrix!! :-)

tonejaco, 2020-04-03:
So Cut, Copy & Paste does work on the Command line of the Console on the Mac too. "Select All" doesn't work. And there is an anomaly about selecting in that command line, it seems the region where that selection works doesn't match the actual size of the command line. It's like it only works in the top half of that region.  Pasting into the command line doesn't want to include any "returns" that are in the copied text. So there are a few things besides copying in the console window that need to be fixed.

tonejaco, 2020-04-03:
More on selection region, it appears you can click anywhere in the Console window and it will insert the cursor into the command line. It shouldn't work that way.

tonejaco, 2020-04-03:
The version uploaded to the forum doesn't have the funny region problem with the Command line in the console.

tonejaco, 2020-04-03:
I changed the title to "Broke" because most of it is there, it just doesn't all work right.

tonejaco, 2020-04-03:
A paste into the command line should probably check syntax and refuse to paste if it doesn't pass. Of course that could be a whole new feature in itself. It should be able to paste multi line gcode commands. (including the line returns)

cocojoe, 2020-04-04:
@tonejaco I had a quick look here, only just noticed Luban source was available 😄  The console functionality is provided by https://github.com/xtermjs/xterm.js 

It's using an older 3.x version, a switch to the new major of this lib 4.x may improve things in general. The issue seems to be dependant on which browser is rendering Luban. If I run this under Windows and use Edge browser, I can use ctrl+c/ctrl+v to successfully copy/paste from the console to notepad. It didn't work under Firefox or Chrome.

https://github.com/xtermjs/xterm.js/issues/2478

tonejaco, 2020-04-05:
Are copying from the display portion of the console window or just the command line portion?

cocojoe, 2020-04-05:
This was for the display portion of the console.

cocojoe, 2020-04-05:
@tonejaco I've raised a PR for this and included a video.

cocojoe, 2020-04-05:
I removed the behaviour of clicking in the console window will autofocus onto the command line. In retrospect, this might have been the root cause issue around copy/paste from the console. As if the focus is switched to the CLI then the window you wanted to copy/paste is now out of focus.

The reason I believe it was done like this is the console window is a fully functional terminal. You can enter commands directly into it. However, that would defeat the point of having the CLI box I presume and it's not as clean in some ways. So that's why it would shift focus from window to CLI box to avoid this happening.

cocojoe, 2020-04-05:
I ran a test on this by allowing the console window to stay in focus and avoid using CLI box.
I could paste in multi-line commands e.g

```
G
M
```

And it works 👍 

The "Feature" fix here, depending on the path the developers want to go down. Is to remove the CLI box and use the console window as it was designed to be used.

tonejaco, 2020-04-05:
Sounds Great!
I like that the up & down arrows work for scrolling to previous commands if you want to re-execute them.

tonejaco, 2020-04-05:
Do you think this is related to the controller not echoing when connected via WiFi? #251

parachvte, 2020-04-10:
With #275 merged, this issues can be closed.

tonejaco, 2020-04-10:
I just noticed this today, something to check if it still exists once this release is out. When the terminal window is expanded to fill the region from top to bottom, the text that was in the Workspace shows through to the console as if the window was tansparent.
[image]

3.3.3 Snapmaker Luban
This on a Mac with 10.14.2 MacOS.

parachvte, 2020-04-11:
It's still there.

tonejaco, 2020-04-11:
Does this warrant a new issue? It's not a copy/paste thing.

parachvte, 2020-04-15:
It's minor issue that was fixed in #290

tonejaco, 2020-04-16:
In 3.5, selecting copy from the Edit menu works, doing a ctrl-C doesn't.

tonejaco, 2020-04-17:
Works ok on my Mac. One thing I noticed on the PC was that when you did the copy command from the menu, the selection went away, that is the highlighting of the selection. Doesn't do that on the Mac.

tonejaco, 2020-05-05:
Time to close.

closed 2020-05-05 by tonejaco

oldztimer, 2021-07-06:
I just updated to the latest Luban 3.15.1 on MacOS 11.4

Copy from the console does not work (again). (I've tried multiple copy key combos: Cmd+C, Ctrl+C, Ctrl+Shft+C)
Paste to the console does not work.

I don't get any contextmenu or app menus to trigger copy/paste by other means.

wri-matt, 2021-07-20:
Confirmed. I am seeing the same results as @oldztimer . No ability to copy from the console window, which makes manually leveling the bed TEDIOUS. No keystrokes, no Edit menu, no context menu. This needs to be re-opened (at least for the Mac).