Snapmaker/Luban #262: M114 doesn't work in a GCODE file.

tonejaco, 2020-03-26:
When I try a M114 in a GCODE file it doesn't show up on the console. I am connected via USB.
At times it seems to trigger a M105 (temperature report) when executed from the console while a GCODE file was executing. I tried a M400 before the M114 but that didn't help. It seems to only work properly when the controller is sitting idle. Definitely a controller issue unless the command isn't making it through the Touch Screen.

tonejaco, 2020-03-28:
Learned something. If you do a M400 just before the M114 then it works, at least when executed thru a Luban Macro.

parachvte, 2020-12-01:
Seems like it's an issue in Luban's communication with the controller, response is sometimes not matched with its request.

sumibin, 2023-03-21:
The issue seems to have been fixed in v4.7. 
[image]

closed 2023-03-21 by sumibin