Snapmaker/Luban #266: Feature Request: Pen Plotter Mode

colinwolff, 2020-03-28:
I'd love a new mode to be added to Luban that enables the use of Snapmaker Original / 2.0 as a pen plotter. Ideally without having to keep a work module installed/connected (Laser or CNC module).

Within the new Pen Plotter mode, you'd be able to generate gcode similar to the Laser engraving mode, and adjust settings, like whether or not a pencil is being using and how fast to press down on the Z axis as it writes. Other settings may also be helpful.

I'm not sure if this would require a firmware update to the Snapmaker Original for this functionality to be supported.

I'd love to hear from others on the interest of this new mode, and any additional settings or features that might need to be considered.

Thanks!

parachvte, 2020-12-01:
A modification on the main controller firmware is required. You need to comment out the security check of tool head module, or add pen plotter as a new module in the firmware.

Please checkout the controller repo (https://github.com/Snapmaker/Snapmaker2-Controller) for further info.
I'm closing the issue for now.

closed 2020-12-01 by parachvte

colinwolff, 2020-12-23:
@parachvte Is the 2.0 firmware (https://github.com/Snapmaker/Snapmaker2-Controller) able to work with the Snapmaker Original? I'm currently using the Original, and I've looked around for a controller repository but haven't found one. Any help is appreciated!

parachvte, 2021-02-07:
Sorry for late response. The firmware repo is only for Snapmaker 2.0 controller.

You can ask our tech support for the source of Snapmaker Original, and make your own modifications.