Snapmaker/Luban #267: No warning on creating more than 10 Luban Macros...

tonejaco, 2020-03-28:
I'm assuming there is a limit to the number of Luban Macros you can create. When I go to create the 11th one it just acts as normal, even letting me save, but no macro. Unless the list is only showing 10. It should put up an error message and not go further.

tonejaco, 2020-03-28:
So perhaps it's just the Macros window not displaying more that 10 macros. When I deleted one, one of the ones created showed up.

parachvte, 2020-03-30:
In `services/api/api-macros.js`, pagination is implemented, but not pagination in frontend page.

tonejaco, 2020-03-30:
Does the window simply need to be able to be enlarged?

parachvte, 2020-03-30:
The existing macro implementation not only has the pagination problem, you can not get execution progress if you set `repeat > 1`. I'd prefer a rework on it and restrict macro limit to 20 or 30.

tonejaco, 2020-03-30:
Sounds reasonable. What is the limit to the size of each macro? Where are they actually being stored?

parachvte, 2020-04-01:
Macros are stored in a JSON file on user root folder (`~/.snapmaker-luban.json` on macOS).
It's not a proper place too, should be stored like other profiles.

parachvte, 2020-04-14:
This issue should be fixed in version >= 3.4.2, the store place is kept unchanged for now.
Can we close this issue? @tonejaco

tonejaco, 2020-04-14:
I think so. Works on my Windows machine.

closed 2020-04-14 by tonejaco