Snapmaker/Luban #274: Extra files (some 55k) are in the 3.3.3 version.

tonejaco, 2020-04-03:
Appear to be in the "Resources/app/node_modules/@jimp" folder on the Mac when looking at the Package contents.

tonejaco, 2020-04-03:
Same for the Windows version. 212 MB worth in the @jimp folder alone.
It must have happened in either 3.3.1 or 2.
3.3.0 didn't have it. 3.3.2 does.
From the look at the size of the installers, I'd say 3.3.1

parachvte, 2020-04-04:
Due to upgrade of `potrace` (2.1.5), a newer version of `core-js` is required.
This leads to two versions of `core-js` are both required (^3.4.1 and ^2.6.5), so each submodule of `@jimp` all requires a copy of `core-js@^3.4.1`. Likely 1800x32 files are added in total.

The solution here is to upgrade the use of `core-js@^2.6.5` which is used by `react-bootstrap@0.32.4`.
We need to migrate bootstrap from v3 to v4.

parachvte, 2020-04-16:
I pre-released version `v3.5.0-rc.0`, number of files now reduced to 9380.

tonejaco, 2020-04-16:
Appears to be fixed in 3.5. I'm only seeing 7,816 files on a Windows 7 64 bit machine.

closed 2020-04-16 by tonejaco