Snapmaker/Luban #285: 2 Problems with 3.4

stgreenb, 2020-04-13:
1. File names for 3.4 release say 3.3.3
2. After installing x64 windows release on win 10, nothing happens when launching the the application (though upon attempting to uninstall, i am prompted that the application is running).  Unstall, try again, same result. Uninstall, go back to 3.3.3, launches fine.

tonejaco, 2020-04-13:
on the Mac it doesn’t work! Runs and only shows a “Snapmaker Luban” menu with a"Quit" menu item.
I tried a restart of the Mac, same thing.

tonejaco, 2020-04-13:
Nothing happens on my Windows 7 machine at work. I see two Snapmaker Luban.exe processes in the Task Manager, One is 6,204k & one is 49,628k. Just look stuck.

parachvte, 2020-04-14:
Sorry I draft the release but I forgot to mark 3.4.0 as pre-release.
Just a build for 3.4.0 testing.

tonejaco, 2020-04-14:
Does it run under Linux?

parachvte, 2020-04-14:
It's a dependency issue related to the DXF, it worked for development environment but I didn't check the production build when merging DXF feature.

Fixing it right now.

parachvte, 2020-04-14:
@stgreenb @tonejaco Check out v3.4.2

parachvte, 2020-04-14:
Closing this issue.

closed 2020-04-14 by parachvte