Snapmaker/Luban #295: Failure of V 3.4.2 to install with Windows 64

heidnerd, 2020-04-15:
No joy with Windows 10 Pro 64bit version 1909,  with windows defender and current patches installed.  Clicking the installer package results in a popup window.  This did not occur with v3.3.3

[image]

More info   click     results in

App: | snapmaker-luban-3.4.2-win-x64.exe
-- | --
Publisher: | Shenzhen Snapmaker Technologies Co., Ltd.

This looks like some kind of cert missing during the build process....

parachvte, 2020-04-15:
Right click to see properties, in tab like "Digital Signature" there are 2 signature for Windows.
Wondering if Windows updated its signature / validation method?

heidnerd, 2020-04-15:
I just checked on the machine that I had tried it on,  other than the
date stamp for the certificate being different I could not see any real
difference.  When I look at the certs both of them said they are okay...
just the April 1, vs April 14 timestamp difference.

There are three zero day exploits that Microsoft is now patching for,  I
will update the PC downstairs and have windows defender update again to
make sure that it isn't just a mismatch on the part of the patterns
defender is using.

I also tried both the V3.3.3 and the new 3.4.2 version again... 3.3.3
pops up and asks for he administrator password to install... 3.4.2 pops
up and says stopped by smart screen.... it is 2AM here so time to head
to bed - I'm repeating myself...  I will try again in the morning.

Anything I can capture in the morning that will help spot the issue?

---

Dennis Heidner

(206) 817-1164 

 

Links:
------
[1] https://github.com/Snapmaker/Luban/issues/295#issuecomment-613790161
[2]
https://github.com/

heidnerd, 2020-04-24:
I've verified that Windows 10 was updated with the latest patches on both the OS and Windows defender as of 4/23/2020.  Same problems happens with the each download version of 3.4.2 windows 64bit,  but I can click on the 3.3.3 version and it will install without a problem..

This isn't just a popup warning.  You can  not get past the block. It stops - no choice to go forward.  You can't get past by trying to install as administrator.

I'm ten km from Microsoft's main campus,  I can could open a ticket with Microsoft - if we can better define what I need to say...

heidnerd, 2020-04-24:
Found the Microsoft explanation that might help others.   Search their support library for
"How to Disable Windows Defender Smart Screen for a particular Desktop Application",  published September 2017.

https://answers.microsoft.com/en-us/insider/forum/all/how-to-disable-windows-defender-smart-screen-for-a/2f67ea6e-b01b-4c78-8820-485a703666a6

I changed the setting from block to warn,  I was able to install the version of 3.4.2

I've confirmed that this version also fixes the bug I also submitted earlier in which trying to look at the layers that the slicer defined -- were all purple (unknown)  that has indeed been fixed.

parachvte, 2020-05-08:
Tested on new installed Windows 10 Pro 64bit, no warnings.

Closing this unless someone reports this issue again.

closed 2020-05-08 by parachvte