Snapmaker/Luban #298: Feature Request: Filament calibration

stgreenb, 2020-04-19:
One of the things I often need to do upon getting new filament is to tweak the settings. In other system, i might do this by building a temperature tower and adjusting the heat (or other settings) at regular Z intervals.  Its one of the things keeping me going back to cura. Adding a modify at Z would work, but maybe a less complex option would work as well. A built in temp tower from the case library where you just enter the lower and upper values?

lichtwellenreiter, 2020-05-28:
@stgreenb: Can you supply any form of Example for the request? Like a Video or an excerpt from the Cura Docs?

stgreenb, 2020-05-28:
@lichtwellenreiter  here's a sample temperature tower that you could build in a few other apps. https://hobbyhoarder.net/temperature-tower/