Snapmaker/Luban #314: Snapmaker Luban 3.30 problem?

blockmodule, 2020-05-06:
Snapmaker fans all over the world
Snapmaker developers

Hello We will report on the following problems than Japan.
I can only use Japanese, so I am using Google Translate to convert to English.
You may be doing the wrong translation.
Please forgive me.

Model: Snapmaker 3-in-1 Original
Firmware: 2.11

https://snapmaker.com/download/snapmaker1
Snapmaker Luban 3.30

1. COM1 in the serial port connection is garbled.
2. There are times when the serial port cannot be recognized.
3. The replacement unit cannot be identified automatically after recognizing the serial port.
With Snapmakerjs, the 3D print head / laser / CNC was automatically identified.
4. The OBJ file cannot be output in G-Code.
The slicer is 0% and the progress bar is 0 and it is not output.

Are you having this problem?
I hope it will be helpful!

-------------------
Japanese Ver

世界中のSnapmaker ファンの皆様
Snapmaker開発者の皆様

こんにちは日本より下記の問題点について報告をいたします。
私は日本語しか使えないため、Google翻訳を使用して英語に変換しています。
間違った翻訳をしているかもしれません。
お許しください。

使用機種：Snapmaker 3-in-1 Original
Firmware：2.11

https://snapmaker.com/download/snapmaker1
Snapmaker Luban 3.30

1.シリアルポート接続のうちCOM1が文字化けしている。
2.シリアルポート認識ができない時がある。
3.シリアルポート認識後に交換ユニットが自動で判別できない。
　Snapmakerjsでは3Dプリントヘッド/レーザー/CNCは自動判別していた。
4.OBJファイルをG-Code出力ができない。
　スライサーが０％でプログレスバーが０のまま出力されない。

皆さんは、このような問題は発生していますか？
参考になればよいのですが！

blockmodule, 2020-05-06:
I made a mistake!
Snapmaker Luban 3.3.3 ...

Don't lose to the coronavirus!
Have a fun Snapmaker life!

---------
Japanese Ver

間違えました！
Snapmaker Luban 3.3.3ですね・・・。

コロナウイルスに負けないで！
楽しいSnapmakerライフを！

blockmodule, 2020-05-08:
I have summarized the contents of the defect in the video.
I'm sorry, but the watermark is included because I used free software.

https://www.blockmodule.com/wp-content/uploads/2020/05/VID_20200508_105225.mp4

blockmodule, 2020-05-08:
I have summarized the contents of the defect in the video.
I'm sorry, but the watermark is included because I used free software.

https://www.blockmodule.com/wp-content/uploads/2020/05/snapmaker.zip

parachvte, 2020-05-14:
Update:

3. For easier maintenance of code, we removed the head detection for Snapmaker Original. Later when we refactor the model selection, we can move the machine model specific code into its definition. Then we can add back the detection.
4. The OBJ slicing problem should be fixed in https://github.com/Snapmaker/Luban/pull/323.

blockmodule, 2020-05-14:
Thank you for contacting us.

1.I understand that the characters assigned to the COM port in the Japanese environment use characters other than UTF-8.

2. The part that fails to recognize the serial port occurs rarely in Snapmakerjs, but it seems to occur less often in Luban.
　This may be a unique problem in my PC environment.

3.If you can, please revert the detection with machine model specific code of 3-in-1 Original(Ver1).
　It will also prevent instruction errors due to human error.

I'm relieved to hear that the slicing problem in the 4.OBJ file will be fixed!

5.I finally understand.
　The Z-Axis has only one physical limit switch and it's a software end stop.
　This is not mentioned in the manual, but it is a prerequisite to run "Home Axes" after booting the machine.
　It is not controllable when the power is turned off, so a natural fall must be a specification.
　I decided to put Styrofoam to prevent the Z-axis from falling when I turn it off:)

closed 2020-12-01 by parachvte