Snapmaker/Luban #315: Z axis goes down without permission

blockmodule, 2020-05-06:
Z axis goes down without permission

Snapmaker fans all over the world
Snapmaker developers

Hello We will report on the following problems than Japan.
I can only use Japanese, so I am using Google Translate to convert it to English.
You may be doing the wrong translation.
Please forgive me.

Model: Snapmaker 3-in-1 Original
Firmware: 2.11

There was no place to define a problem with the machine itself, so I'll report it here.

・ When the machine power is on, press the Z-axis lightly to lower it.
・ If the Z-axis is pressed down lightly when the power is off, it will go down to the bottom.

This condition has also been reported by other buyers in Amazon Japan's buyer comments.

The nozzle of the 3D printer and the base come into contact, so please improve it.

----
Japanese Ver

Z軸が勝手に下がっていく

世界中のSnapmaker ファンの皆様
Snapmaker開発者の皆様

こんにちは日本より下記の問題点について報告をいたします。
私は日本語しか使えないため、Google翻訳を使用して英語に変換しています。
間違った翻訳をしているかもしれません。
お許しください。

使用機種：Snapmaker 3-in-1 Original
Firmware：2.11

マシン本体の問題について定義できる場所がなかったので、こちらで報告いたします。

・マシン電源が入っている状態でZ軸を下に軽く押すと、自然と下がる。
・電源が入っていない状態でZ軸を下に軽く押すと、一番下まで下がってしまう。

この状態は他の購入者がAmazon Japanの購入者コメントにも報告されています。

3Dプリンタのノズルとベースが接触するため、改善をお願いします。

parachvte, 2020-05-07:
You can use M17 / M18 / M84 G-code commands to enable/disable steppers when powered on, it can prevent the nozzle from sliding down.

Or you can contact support@snapmaker.com for help.

blockmodule, 2020-05-08:
Dear parachute

Thank you for your reply
I tried it but didn't get the expected results.
I will investigate variously.

closed 2020-12-01 by parachvte