Snapmaker/Luban #316: DXF Arc 360

WaldoMinny, 2020-05-07:
I've identified that Arcs that start before 360 degrees and go through 360 to end after 0 degrees are not imported.  There are also small arcs that have an issue.
In the attached drawing, there are;
362 Arcs
94 Circles
13 Elipses
2457 Lines
1 Polyline
4 Splines
Attached is also a screenshot with some elements in red that don't load in Luban 3.4.2.  In the Luban screenshot, you can see the arcs not being drawn as well as some other arcs that are not fully drawn in blue.  These are not the ones going through 360, they are merely drawn halfway.

Please feel free to contact me with any additional drawings or tests.

[image]
[Millenium Falcon Rev 17 - Thin Engraving.zip](https://github.com/Snapmaker/Luban/files/4595683/Millenium.Falcon.Rev.17.-.Thin.Engraving.zip)

jiantaolovebingbing, 2020-05-11:
Yes, I did not consider that the startangle may be bigger than the endangle.I fixed it , you can see the result in the screenshot. But there are a bit problems in some arc, maybe you can help check the startangle and endangle of these 4 'arc'.Thanks!
[image]

jiantaolovebingbing, 2020-05-11:
Yeah, I did not consider that the startangle may be bigger than the endangle.I fixed it , you can see the result in the screenshot. But there are some extra line  in some arc, maybe you can help check these 4 arc's data. Thanks!
[image]

jiantaolovebingbing, 2020-05-11:
Yeah, I did not consider that the startangle may be bigger than the endangle.I fixed it , you can see the result in the screenshot. Maybe you can check if there are other missing line?
[image]

WaldoMinny, 2020-05-11:
Hi Jiantao

Thank you very much for your assistance.  I think some of the smaller lines
come from other parts of the drawing that are drawn at the wrong location.
How would you like me to test this?  Do you have a compiled version that I
can test?

Regards
Waldo Minny
waldo@minny.com
+27 82 901-2384

WaldoMinny, 2020-05-11:
> Yeah, I did not consider that the startangle may be bigger than the endangle.I fixed it , you can see the result in the screenshot. Maybe you can check if there are other missing line?
> [image]

Thank you very much

jiantaolovebingbing, 2020-05-12:
mac or windows?

jiantaolovebingbing, 2020-05-12:
https://github.com/Snapmaker/Luban/releases/tag/3.5.0-rc.1

parachvte, 2020-06-03:
Closing this issue as corresponding PR was merged.

closed 2020-06-03 by parachvte