Snapmaker/Luban #362: Feature Request

msteelepmp, 2020-06-27:
Integrate with OctoPrint (aka Octopi). There is an interface to upload direct from Cura, but in many ways I find Luban easier to use...if only there was a way to send the GCODE files directly from Luban on my Mac to my Octopi controlling my Snapmaker Original.

tonejaco, 2020-07-02:
You should really update the title of this request to actually SAY what the request is.
i.e. "Feature Request: Ability to connect to OctoPrint".

What I would rather see is Luban updated to completely negate the need for OctoPrint. Do all the essential things that OctoPrint does right in Luban.

parachvte, 2020-07-03:
I'd like to have Luban running on Raspberry Pi as well, which needs us to build image for Pi and port frequently used functions and plugins to Luban. The later one is what we interested as we intended to run plugin system and make Luban extensive.

There are lots of work to be done before we make it really handful.
What you requested is a quick solution to use pre-existing OctoPi's functions, very reasonable. We are not heavy users of OctoPi so it's relatively low priority to us, we would love to have the community to write a widget to the work rather than waiting in the request queue.

kbaegis, 2021-01-19:
I own a snapmaker as well as a cnc machine. I’d rather be able to connect to a raspberry pi connected to my cnc to send gcode files directly using the same software I use to connect to my snapmaker.

One half of this falls on this project- namely being able to transmit projects to a raspberry pi running octoprint, and the other is asking for an octoprint enhancement to support whatever discovery protocol Luban is leveraging to detect their printers.

I’d vastly prefer a wired interface (eth/tcp/ip >>> serial) over wifi.