Snapmaker/Luban #368: G1 E100 removes the filament

highcast, 2020-07-01:
I am trying to calibrate the A350. 

In Luban 3.6.1, I set the feed rate which after the Wifi disconnected (this is normal here that it goes off after few moments each time). Reconnected and then trying to extrude 10 mm of filament. Nozzle is set to 200 degrees Celcious. This is Snapmaker white PLA. 

[image]

After G2 E100, the extruder RUNS BACKWARDS and removes the filament from the nozzle. 
Running 1.9.0 Firmware.

parachvte, 2020-07-06:
Did you use G1 E before, in G90 (Absolute Positioning) use a E less than previous E will lead to extraction.
You can use G91 (Relative Positioning) instead.

highcast, 2020-07-09:
Entering G9x code helped to this. Thus, I upgraded to 3.7.0 between posting
the issue and your message.
It seems the device was underextruding quite a bit. (demanded: 100 mm,
extruded: 83mm)
But that's another thing, not about this.

closed 2020-08-03 by parachvte