Snapmaker/Luban #371: Missing 3D panel

EcuadorGarcia, 2020-07-03:
I have a workstation where Luban is working perfectly, due to COVID, I got a laptop and installed Snapmaker software, and it's missing the 3D panel, it's only Showing CNC and Laser
no matter what version I install, it's still missing
what can I do?

parachvte, 2020-07-06:
Are you using a Win32 system?
Cura do not provide 32bit version from version 3.0, we hide the tab in that case.

EcuadorGarcia, 2020-07-06:
Yes. It's 32bit

What can I do??

parachvte, 2020-07-06:
Old Luban versions (around Snapmakerjs v2.5.0) provide 32-bit support, or you may need another 3D slicer.

parachvte, 2020-08-03:
Checkout the link below if you want an old Luban version that supports Windows 32 bits.

https://github.com/Snapmaker/Luban/releases/tag/v2.5.2-win32.1

closed 2020-08-03 by parachvte