Snapmaker/Luban #372: Optimize Path does not work

Abolfazl, 2020-07-04:
When I check `Optimize Path` when loading a svg file, it does not seem to do anything. The output file is the _exact_ same as a file that did not have `Optimize Path` checked.

This seems to be the case with any .svg I load.

I am running V3.7.0 stable

parachvte, 2020-07-06:
Seems like we introduce a new bug.

parachvte, 2020-07-20:
https://github.com/Snapmaker/Luban/pull/388

Abolfazl, 2020-07-23:
@parachvte Why does this fix limit optimization at 2000 shapes? https://github.com/Snapmaker/Luban/pull/388/commits/5d921f0a5dec1b64353f70ad9f3d8a82106b69dc

parachvte, 2020-07-27:
Optimize Path uses an unoptimized short path algorithm, which is at O(N^2) complexity. With number of shapes grows, it will be very slow.

Abolfazl, 2020-07-27:
Is there a way to optimize Optimize Path? I do not mind the added time

parachvte, 2020-08-04:
@Abolfazl 
> I do not mind the added time

We still need it to complete under a limited time (in 1 second or in a few seconds).
I come up with some ideas to improve it, I'll put this into our backlog.

siberianhuskies, 2022-05-18:
Instead of using an exact short path algorithm, just selecting greedily the next nearest path/point, should already provide a sufficient result.

Currently when cutting something with the laser, there are still lots of unnecessary movements from "left to right". If one just selects the nearest path to continue cutting, one could save a lot of travel movements. Yes, this is not the optimal path, with the overall shortest travel movements. However, a good approxmiation which should result in 90% or 95% less travel movements, which should be good enough for most use cases while still being fast (and easy to implement).

sumibin, 2023-03-21:
The new path planning algorithm shows better results and subsequent issues can be proposed on this basis.
[image]

closed 2023-03-21 by sumibin