Snapmaker/Luban #380: Model on its side

nblount, 2020-07-12:
Luban 3.7 exporting to G-Code all looks fine but when you load the G-Code onto the Snapmaker 2.0 the model will be on its side. Generating the same code on 3.6.1 does not have this problem.

parachvte, 2020-07-15:
What do you mean by "on its side"?

nblount, 2020-07-15:
When you upload the model onto the Snapmaker it displays an image of what is going to be printed on the controller The model shows on its side (90 degrees off horizontal). If the same model is converted to g-code on older versions of Luban it displays correctly. This could be cosmetic but I have been unwilling to let it print something that doesn't display correctly on the controller.

parachvte, 2020-07-20:
I got it.

parachvte, 2020-08-03:
Fixed in v3.8.0

closed 2020-08-03 by parachvte