Snapmaker/Luban #386: CNC mode crashing when generating tool path

fvlasie, 2020-07-17:
I ma trying to generate G-code for the CNC function using a depth map. Luban keeps crashing when I press "Preview" to generate  the tool path.

MacOS 10.15.5 Luban 3.7.0.331

Thank you!

fvlasie, 2020-07-17:
[image]
[image]

fvlasie, 2020-07-17:
[image]
[image]

fvlasie, 2020-07-17:
I also tested this on Luban 3.7.0 on Windows — same problem. :(

fvlasie, 2020-07-17:
The same problem occurs on CentOS 8.

fvlasie, 2020-07-17:
Also tested with versions 3.5 and 3.6.1—same issue

fvlasie, 2020-07-17:
Maybe for complex projects we could have the option of generating G-code without the preview?

fvlasie, 2020-08-02:
The problem still exists with version 3.8.0

parachvte, 2020-12-01:
Is it still occurs in v3.11.0 or v3.12.0?

closed 2020-12-01 by parachvte

fvlasie, 2020-12-03:
I just tried it in 3.12.2 and the problem is fixed! Yay!
Thank you very much!
谢谢