Snapmaker/Luban #402: 3.8.0 does not save custom print profiles

highcast, 2020-07-29:
Customized print quality profiles do not get saved. When closing the app, those are lost (except if you export them) and you need to do it again upon next program usage. 

Windows platform
Luban 3.8.0

parachvte, 2020-08-03:
I can not re-produce this.

Check if you are still using the right profile, or if it's the write permission issue of userData folder.

highcast, 2020-08-18:
@parachvte here's a video record of this issue happening. 

[](https://youtu.be/q9aAZM92cnM)

 or if it's the write permission issue of userData folder.
- What is the specified userdata folder in Windows 10 profile? Where is it?

YDKK, 2020-08-30:
I have the same issue on Windows platform.
I had upgrade from 3.7.0 to 3.8.0 then all print profiles are lost and Luban entirely no longer save new print profiles.

wdecook, 2020-08-30:
I have the same issue on Windows 10 after updating to Luban 3.8.0

wdecook, 2020-08-30:
@parachvte Where is the userData folder? I can't find this directory anywhere in the program file structure.

Pflugshaupt, 2020-09-05:
I am facing the same issue on multiple windows 10 machines. I started out with 3.8 without prior installation. Material settings do work, but custom print settings are gone on relaunch and it is quite annoying. Reimporting from json also does not work.

epic709, 2020-09-11:
same issue encountered on Win64 machine. Have double checked all file permissions and nothing works. Can only save print settings to JSON file but unable to load back into Luban. All custom profiles disappear after restarting Luban

usefool, 2020-09-24:
I have the same issues with Luban 3.8.0 on my Mac. 
1) A printer profile is forgotten after restart of Luban
2) Saving a JSON file works but loading a Json file has no effect

usefool, 2020-09-25:
Is fixed on Mac with 3.9

closed 2020-12-01 by parachvte