Snapmaker/Luban #412: Luban 3.8 Snap Maker Laser G-code Generator Save files issues

Letho-BFG, 2020-08-12:
Dear reader,

I am happily testing our new Snap Maker 2 A350.
We've got some nice tests and results in Luban.

I am a leatherworker and have been educated in machining.
So I wanted to help myself and other people out by making a benchmark table for laser power with work speeds.

Currently in Luban every entry gets it's own G-code.
So if I load a DXF with my set up I can not change the individual units/shapes.
A fix for that is to load all my units/shapes individually (all 162 of them)
And manually modify their parameters (Laser power %, Work speed, other settings to uncheck).

After a lot of hours and hard work I finished the table.
During the process I saved multiple times via the save (ctrl +s) and Save as button. This created a .SNAPLZR file.

I have found multiple issues with my current Luban version (3.8.0):

- If I save in the Laser suite of Luban without generating the G-code the file will be empty (91bytes)

- If you have generated the G-code and saved. When you reopen the file (currently this only works when you have opened Luban and the go to 'open' when you are in the Laser suite.) The file gets regenerated and plotted within the laser program.
But all the parameters are set to default! (laser %, work speed, fill etc.)

- I can not select multiple units/shapes in the laser suite. This means that when I need to adjust the size of my table I have to manually reposition all 180 unit shapes.

- When selecting a unit/shape you can easy click another and move it. Is there a lock XYZ? I would love that!

- The generation of the G-code feels opposed to what the guide video of the official Snap Maker team suggests.
It states that the order of loading in units/shape is the order of printing. If you want to have more control use the slider 'print order' (1-10) "If you leave the units/shapes on default the loading order that will be the printing order."
But every time I get a scramble of the lines and shapes. See attached files.

Is there any thing I can do to help? Or do you require more information?

Kind regards Letho (BFG)

[image]
[image]
[Snap Maker A350 Laser Benchmark table SNAPLSR.zip](https://github.com/Snapmaker/Luban/files/5063220/Snap.Maker.A350.Laser.Benchmark.table.SNAPLSR.zip)
[Template Snapmaker Laser diagr_54314024 NC.zip](https://github.com/Snapmaker/Luban/files/5063221/Template.Snapmaker.Laser.diagr_54314024.NC.zip)

Wynand-S, 2020-08-28:
### **WRT the laser (and CNC) in Snapmaker Luban 3.8.0**

I've found that Luban can't handle SVG (vector) files with multiple layers. On the "editor" side it shows the correct 'visible' layer, but as soon as you move to the "Process" side the full SVG with all layers including hidden layers are present and gets exported to the Workspace and thus also gets engraved/cut ---> See attached screenshots of the same file (Editor vs. Process)
There is sadly a lot of hard work that needs to go into the editor/processes side of Luban, and I'd love to see it improve by helping with issues and suggestions!
We need :

1. Alignment options between files (eg. Centring horizontally or vertically, in relation to the other selected objects or the origin etc.
2. Treating multiple layers of the Vector files as different tool paths will really help bring Luban closer to a bare minimum editor.
3. Ability to Group / Ungroup objects to allow for more effective duplication and allowing you to output multiple copies onto your work surface quickly instead of the manual tedious method currently required.
4. Use of different colours to differentiate between toolpaths (work speed/multi-pass/cut/etch) etc - use multi-layer feature in SVG files to create the various toolpaths in one file
5. Luban can't even re-open it's own saved project files without freezing up - it's imperative that hard work that has been saved isn't lost!

jane-rose, 2020-09-13:
I will try to reproduce your operation tomorrow and reply to you. Thanks!

jane-rose, 2020-09-16:
hi,
For the first big issue:  individual units/shapes, we will do it when  laser/cnc function' new feature is stable.But it takes time.
For point 1, we can't  re-produce it.
Point 2, it does exit. And we will start it rigth now on next version 3.10.0.
Point 3, this is on the next 2 or 3 version schedule.
Point 4, version 3.9.0 will have uniform scale function.
Point 5,  it works fine. Once u change the order, you have to press 'enter' button or click other place in widget to make the change work.
Thanks!

Letho-BFG, 2020-09-16:
Jane,

Thank you for looking into the problems I faced.
I wish you all the ebay of luck for the expansion of Luban!
Keep up the great work!

*Kind Regards,*Letho Veldt

lethoveldt@gmail.com

bruno-71, 2020-11-02:
Hi @jane-rose ,

I saw above that you had mentioned fixing the issue of the laser settings not being saved.  However, it still does not seem to be fixed in the latest 3.10.2 version.  Any update on when this will be fixed?  It is very tedious to have to enter all the settings every time I open a file.

From the original post:

> If you have generated the G-code and saved. When you reopen the file (currently this only works when you have opened Luban and the go to 'open' when you are in the Laser suite.) The file gets regenerated and plotted within the laser program.
But all the parameters are set to default! (laser %, work speed, fill etc.)

Your reply:

> Point 2, it does exit. And we will start it rigth now on next version 3.10.0.

closed 2023-03-21 by sumibin