Snapmaker/Luban #419: how to build the application

iceshu, 2020-08-25:
I git clone the project and run dev is ok, but I want to build the application run the following

```js
    "build:mac-x64": "bash -c 'build/electron-builder.sh --mac --x64'",
    "build:linux-ia32": "bash -c 'build/electron-builder.sh --linux --ia32'",
    "build:linux-x64": "bash -c 'build/electron-builder.sh --linux --x64'",
    "build:win-ia32": "bash -c 'build/electron-builder.sh --win --ia32'",
    "build:win-x64": "bash -c 'build/electron-builder.sh --win --x64'",
````

I run these commend but error with  
```bash

'electron-rebuild' is not recognized as an internal or external command,
operable program or batch file.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! snapmaker-luban@3.8.0 electron-rebuild: `electron-rebuild "--version=" "v2.0.18" "--module-dir=dist/Snapmakerjs" "--which-module=serialport"`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the snapmaker-luban@3.8.0 electron-rebuild script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

and i install 'electron-rebuild' but not work

parachvte, 2020-08-25:
Install dependencies using `npm install` (make sure you have `electron-rebuild` in `node_modules` folder).

And then `npm run build` and `npm run build:mac-x64` (assume you're using macOS).

iceshu, 2020-08-25:
@parachvte  thks，I try it angin and it seems ok，but still not work， I think error with network ，
[image]

After all I am in china。the firewall is too strong~

frank-pian, 2020-08-25:
> 
> 
> @parachvte thks，I try it angin and it seems ok，but still not work， I think error with network ，
> [image]
> 
> After all I am in china。the firewall is too strong~

挂代理，在国内没办法

closed 2020-08-26 by iceshu