Snapmaker/Luban #422: Luban 3.8.0 Laser Suggestions + Editor/Process & File-save issues

Wynand-S, 2020-08-28:
### **WRT the laser (and CNC) in Snapmaker Luban 3.8.0**

I've found that Luban can't handle SVG (vector) files with multiple layers. On the "editor" side it shows the correct 'visible' layer, but as soon as you move to the "Process" side the full SVG with all layers including hidden layers are present and gets exported to the Workspace and thus also gets engraved/cut ---> See attached screenshots of the same file (Editor vs. Process)
There is sadly a lot of hard work that needs to go into the editor/processes side of Luban, and I'd love to see it improve by helping with issues and suggestions!

**Suggestions / Issues:**
1. Alignment options between files (eg. Centring horizontally or vertically, in relation to the other selected objects or the origin etc.

2. Treating multiple layers of the Vector files as different tool paths will really help bring Luban closer to a bare minimum editor.

3. Ability to Group / Ungroup objects to allow for more effective duplication and allowing you to output multiple copies onto your work surface quickly instead of the manual tedious method currently required.

4. Use of different colours to differentiate between toolpaths (work speed/multi-pass/cut/etch) etc - use multi-layer feature in SVG files to create the various toolpaths in one file

5. Luban can't even re-open it's own saved project files without freezing up - it's imperative that hard work that has been saved isn't lost!

Thank you for your consideration!

jane-rose, 2020-09-13:
As far as I know, the third and fifth points will appear in the next version.  Plus, you can use shortcut keys such as ctrl+c and ctrl+v in version 3.9.0.
Other points may be on the agenda. Thanks!

Wynand-S, 2020-09-13:
Awesome!

Looking forward to an Update :-)

Hopefully we will see multiple object alignment (left / center / right /
vertical / middle / horizontal) as an option too!

Thanks
 Wynand

sumibin, 2023-03-21:
https://github.com/Snapmaker/Luban/issues/2053
The issue is tracked in other issues.

closed 2023-03-21 by sumibin