Snapmaker/Luban #423: [Feature Request] Generate RPMs for releases

TurboTurtle, 2020-08-29:
I use Fedora as my daily driver, not unlike a lot of other people in the maker space. It'd be extremely helpful to produce release RPMs alongside the already produced DEBs.

If desired, I would be willing to package this for the Fedora repos (assuming it is accepted, I don't see anything immediately that would prevent it). I do not know Javascript so I would not be able to contribute code fixes, however I already maintain a few packages for Fedora and am familiar with the packaging processes.

parachvte, 2020-09-24:
@TurboTurtle Please checkout v3.9.0 .rpm release.

TurboTurtle, 2020-09-24:
Installed on F32 without issue, and looks to be functioning correctly. \o/

Thanks for jumping on this!

closed 2020-09-25 by parachvte