Snapmaker/Luban #478: Lucan crashes when using large files in CBC.

gysnst2, 2020-10-10:
Luban crashes all the time for me.  It has become a program that i cannot use.

jane-rose, 2020-10-12:
Can you describe it in more detail and provide the file?

gysnst2, 2020-10-13:
Yes when i load and file into luban and then increase the size to larger then 250x250 the program crashes.  I will create a video and submit this.

Havenissens, 2020-12-16:
My luban crashes by adding 6 of this minifig 
It uses up 95% of my 16Gb ram, while preparing the slice

[Gr-10-Barovian Witch.zip](https://github.com/Snapmaker/Luban/files/5705628/Gr-10-Barovian.Witch.zip)

gysnst2, 2021-04-28:
LuBan seams to no be crashing for me anymore.  3.15 seams to be working well.  It's slow at slicing in cnc but it's working.  Importing an stl into cnc give me the wrong orientation

closed 2022-12-22 by parachvte