Snapmaker/Luban #479: Feature request add sal to luban cnc

gysnst2, 2020-10-10:
I would love to see import of stl into luban. It seams this should be a no brained considering this is also a 3D printer and marketed to the 3D printing community.

jane-rose, 2020-10-12:
It is on the agenda, thanks.

gysnst2, 2021-04-28:
In 3.15 when I import an stl into the cnc slicer the orientation is incorrect and their is not a way to fix it.  [image][image]But if I import that same model into the 3D printer slicer it is imported correctly. [image]

parachvte, 2021-04-28:
[image]

You can change "Projection Orientation" of object in the right side panel.

closed 2021-04-28 by parachvte