Snapmaker/Luban #485: Luban hangs when adding gcode

halr9000, 2020-10-19:
I have one specific gcode file which was sliced using Cura 4.7.1. When I go to workspace and try to load it using "open gcode file", the display shows an empty white window and nothing else but the menu at the top (file, view, window options).

OS: Linux
Luban 3.9.0

Github will not allow gcode uploads, this link should work: https://drive.google.com/file/d/1KbeQhHZdaGtjiDh6y19gzJY9FUQxRRjj/view?usp=sharing

halr9000, 2020-10-19:
[image]

jane-rose, 2020-10-19:
The file you uploaded is not a typical gcode file, please check it!

closed 2020-10-19 by halr9000