Snapmaker/Luban #502: .svg Files imported from InkScape have the wrong size.

Pflugshaupt, 2020-10-25:
Since version 3.9.0 and also in the 3.10.0 alpha, svg files using mm units coming from InkScape get much too small (about 4x) when loaded into Luban to laser cut. It was fine in 3.8.0.

thisismlh, 2020-11-09:
Same is true for Adobe Illustrator

t-foster, 2020-11-15:
Same with svg files exported from openscad.

openscad code:

    square([200,40]);

then render and export to svg produces the following svg xml
```
<?xml version="1.0" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg width="200mm" height="40mm" viewBox="0 -40 200 40" xmlns="http://www.w3.org/2000/svg" version="1.1">
<title>OpenSCAD Model</title>
<path d="
M 0,-0 L 200,-0 L 200,-40 L 0,-40 z
" stroke="black" fill="lightgray" stroke-width="0.5"/>
</svg>
```

Then when it gets imported into Luban cnc and laser it ends up being 25.6 mm wide and 5.1 mm tall.  Maybe the SVGs are just getting bound down to 1 inch at the greatest distance?

t-foster, 2020-11-15:
Although as a user I can then just peek at the first few lines of the svg xml and take the width/height values and after adding to luban manually set the size of my drawn object.  IDK if illustrator or inkscape have those xml fields in them but it may be worth checking.

Pflugshaupt, 2020-11-15:
Of course they have them. This is just a silly bug in the svg importer - hope it gets fixed soon.

t-foster, 2020-11-18:
Sometimes you need to spoon-feed a developer a test case and what you expect for a result so they can put that into some sort of unit test to make more resilient code.

srfin, 2020-12-02:
It appears that all laser file imports are too small? I just opened the "Laser cut gift box" test project on 3.12.3 and it was only 25.6x25.6mm. I also imported a dxf I had created myself and it was also much too small.

srfin, 2020-12-02:
I just did some testing of old versions and found that 3.8 is the last version which correctly loads the gift box project. Unfortunately that isn't able to correctly load DXF files from Fusion 360.

Looks like I'll just give up on laser cutting precise dimensions until this is fixed.

Pflugshaupt, 2020-12-02:
Indeed.. using 3.8. because of that  which annoyingly cannot store 3d print custom profiles :(

jane-rose, 2020-12-03:
@Pflugshaupt  Is it occurs in v3.12.2? For storing 3d print custom profiles, this feature should be available in version 3.12.2

Pflugshaupt, 2020-12-03:
profiles is fine since 3.9, but I need laser cuts that have the right size so I never went past 3.8. I am waiting for this issue here #502 to be fixed.

patchrocks, 2020-12-09:
I just ran into the same (or similar) issue: I use FreeCAD 0.19 and Luban 3.12.3. When creating a SVG or DXF, the model is smaller when importing in Snapmaker Luban (a circle with 20 mm diameter results in a circle with 2,8 mm in Luban). Is this planned to be fixed in a future version, or is there any workaround - or something I do wrong? Thanks a lot!

patchrocks, 2020-12-11:
Additional information: I can confirm that version 3.8 works as expected. Scale is correct when I load the SVG from FreeCAD in Luban 3.8.

dwienie, 2020-12-29:
I installed 3.13.0 and I have also an issue importing svg file in luban.  It is to small but indeed you can resize but the bigger problem is that luban adds a whitespace around the object even if the source svg file has zero margin.   This gives issues in resizing.   you can not scale it to a size that is exactly the size of the source because of that margin.    I tried with DXF, SVG and PNG (in Vector mode) all of them have a margin around the object.  Please fix this otherwise we can not cut  precise based on the source object.

Vincent-Noben, 2020-12-29:
Some tests with inkscape en illustrator.
Weird thing is that it does look normal in "Process" mode, but bad in Edit mode. (I set width to 200mm)
I currently cannot test it but now I wonder if it is only a visual bug of the edit mode, or if the actual lasered result is also to small.

https://github.com/Snapmaker/Luban/issues/502
[image]

parachvte, 2020-12-30:
It's a bug introduced in v3.9.0, we are still refactor the SVG editor to fix it.
You can trust the "Process" result rather than in "Edit".

Galindorf, 2021-01-31:
Can confirm, Luban 3.13.1 importing .SVG for laser cutting, files saved in CorelDraw. Imports the entire file about an inch in size. I'm glad to know this is a bug, I am a brand new user and I thought this was a terrible program!

Pflugshaupt, 2021-01-31:
Seeing that I reported this bug in October and it is still not fixed does make me feel it is a terrible program :(. It is easily the weakest point of the whole snapmaker package. The slicer is missing so many (cura) features, the cnc part is mostly a joke and lasering is crippled by this very bug that feels like it could be fixed in a few hours.

I wished they would just abort this project and instead provide first class support for Cura/Fusion 360 and some laser engraving software. Luban wants to do too many things at the same time and fails at all of them in some way.

srfin, 2021-02-01:
I think its pretty clear that Snapmaker doesn't have the resources to give software the attention it needs either on Luban or the firmware. The pace of development is very slow even for important bug fixes like this. It would be great if they would partner with 3rd parties to provide licenses/support for other programs.

I'm really interested in the rotary module, but I can't see how they're going to provide the software needed to use it well. I'm definitely not interested in paying full price to get 4-axis support in Fusion 360 for example.

marsiancba, 2021-02-04:
Around 1 month ago I have submitted a pull request for this:

https://github.com/Snapmaker/Luban/pull/681

Not sure why it is not merged/fixed already - it is just 1 line change.

t-foster, 2021-02-04:
> Not sure why it is not merged/fixed already - it is just 1 line change.

I'm not one of their programmers and haven't explored their source code enough to tell if it is set up for unit testing, but speaking as a programmer in general I wouldn't take a one line change without unit tests for that area which break without the change and pass with it.

parachvte, 2021-03-01:
Updates on PR #681:

We have #773 and #778 pull requests to solve this issue, they were merged and will be released with v3.15.0.

While there are still several bugs regarding the SVG editor, we are working on them currently.
I'll keep this issue open until release of v3.15 (about April 6th).

Mortisaga, 2021-03-12:
I don't know if it is the same issue I have or at least related:
when adding files to Luban that I made with inkscape, not only it appears way to small but when I resize it to my original size it will have a strange border around... (see pic)
[image]

the original vector is 107 mm in height, but in Luban it sizes the border to 107 and my object is way too small...
I tested this with 3.13.1 as well as 3.14.0 alpha 14 - same problem
I found this very problem with .dfx, .svg, .png files (tested no others)

parachvte, 2021-04-27:
Please check if the issue still occurs in v3.15.0.

I'm closing this issue for now.

closed 2021-04-27 by parachvte

hotchips, 2021-10-31:
Was having same issue - Worked when I loaded original SVG into illustrator and re-saved with SVG 1.0 spec

faberc, 2023-12-17:
This problem exists in Luban 4.10.1

LMalyurek, 2023-12-23:
Confirm with faberc.  Scale still not coming in correctly with 4.10.1.  Please put some effort into this, accuracy matters to almost everything I want to do on this machine.  I am honestly amazed that this has been apparently broken for so long....

Juntrimp, 2024-03-27:
Still an issue in 4.11.0