Snapmaker/Luban #509: v3.9.0 Text options lost in the laser/CNC G-code Generator

TurboPunk, 2020-10-26:
In the laser/CNC G-code Generator text object used to have more options in v3.8.0

Steps to reproduce:
Add a text object in v3.9.0 in the laser/CNC G-code Generator and inspect the properties.
Enter multiple lines of text.

Missing:
- Line height
- Allignment

No longer working:
- Multiline text

Screenshot v3.8.0:
[image]

Screenshot v3.9.0:
[image]

parachvte, 2020-10-27:
Yes we rework the text function to support WYSIWYG, so the alignment options were removed temporarily...

While the text is still buggy, we will bring back these options when the text function is stable.

sumibin, 2022-03-24:
[image]
v4.1.4 solution

closed 2022-03-24 by sumibin