Snapmaker/Luban #534: File save does not save laser process settings

bruno-71, 2020-11-05:
Hi,

When saving a file in Luban, it does not save the laser process settings such as print order, jog speed, work speed, passes, pass depth, and power.  Every time the file is opened, all the settings are reset back to their default values.  This makes it impossible to setup projects for printing/cutting later.  The values have to be typed in for all layers every time.  This is a very significant problem.

Thanks.

MiracleMike, 2020-11-06:
Hi, 

Same thing happens for the CNC settings. 
I have 3 shapes in the .snapcnc file and all 3 bits & configs are reseted to defaults when I reopen the file.

Thanks

Happy1Snappy, 2020-11-07:
Hi
same Problem here, i use Version 3.10.2 on Mac and Windows10 64Bit . Is this Bug in all Versions ?
Thx

jane-rose, 2020-11-16:
Sorry, at the time ,  these parameters have not been added to the project file.

bruno-71, 2020-11-17:
In issue #412  you mentioned it would be fixed in version 3.10...
https://github.com/Snapmaker/Luban/issues/412#issuecomment-693134335

Do you have an update of when this will be fixed?  There is no reason to save project files if it doesn't save the settings.

jane-rose, 2020-11-18:
Sorry for that, when we sorted out the project files, we found that there are many things that should be put into the project files. I will push it as soon as possible, any progress will update in this issue. Thanks!

> In issue #412 you mentioned it would be fixed in version 3.10...
> [#412 (comment)](https://github.com/Snapmaker/Luban/issues/412#issuecomment-693134335)
> 
> Do you have an update of when this will be fixed? There is no reason to save project files if it doesn't save the settings.

Sorry for that, when we sorted out the project files, we found that there are many things should be put into the project files. 
I will push it as soon as possible, any progress will update in this issue. Thanks!

patchrocks, 2020-12-27:
+1
This feature would be a great time saver and would make life a lot easier.

afewell, 2021-02-24:
+1 any updates on this? Loving the 2.0 but the inability to save settings and presets is really unfortunate considering the user experience is great overall, but lack of saving setting is a huge huge problem for the user experience. Thanks!

jane-rose, 2021-02-25:
Yeap,it will be released on version v3.15.0.

closed 2023-03-21 by sumibin