Snapmaker/Luban #535: Feature Request: allow automatic wi-fi connection

kps1ny, 2020-11-05:
Please can you add an option to the Snapmaker controller UI that allows automatic wi-fi connections from Luban? Our Snapmaker is in a different room from our computers running Luban and it's inconvenient to have to go over to the device just to tap the screen to accept the wi-fi connection from the computer, then go back to the computer and send the file. We actually have multiple computers running Luban, so having the controller allow all connections would be most helpful. Thanks for your consideration.

mr-manuel, 2020-11-12:
We alse have the Snapmaker in another room. It would also be nice to see the progress on the computer.

kps1ny, 2020-11-12:
> We alse have the Snapmaker in another room. It would also be nice to see the progress on the computer.

This can already be done. If you submit the job via wi-fi, Luban will report on its progress. My feature request has to do with the initial file transmission, which requires tapping the printer interface to confirm the connection. That step is what I'd like to be able to turn off.

mr-manuel, 2020-11-12:
> This can already be done. If you submit the job via wi-fi, Luban will report on its progress. My feature request has to do with the initial file transmission, which requires tapping the printer interface to confirm the connection. That step is what I'd like to be able to turn off.
Ok, thanks for the info. In my case Luban disconnects after the job is transmitted, so I thought this is normal.

Connecting with a PIN or IP whitelist, that don‘t request the manual confirmation on the Snapmakers touch screen, would maybe be a nice solution.

Airmonkdev, 2020-11-15:
"Connecting with a PIN or IP whitelist, that don‘t request the manual confirmation on the Snapmakers touch screen, would maybe be a nice solution."

I agree! Obviously we can't accept that everyone can get connected to our Snap without some security confirmation but remember the already confirmed IP address or something like mr-manuel suggested would be very nice to not touch the screen every time for "trusted request". 

Thanks and keep up the good work!

jane-rose, 2020-11-16:
> "Connecting with a PIN or IP whitelist, that don‘t request the manual confirmation on the Snapmakers touch screen, would maybe be a nice solution."
>
> I agree! Obviously we can't accept that everyone [...]

cool, that may be a option

patchrocks, 2020-12-27:
+1

Same situation for me. The Snapmaker is in a different room and I simply want to transfer the G-code from my computer to Snapmaker. In my case it would also be ok to not even use a PIN code as one anyway needs to be in my home network to access Snapmaker (maybe that could be a configuration option). PIN code or IP would also be great.

jenkins007, 2021-01-12:
Same for me. I have to walk through 3 different rooms to get to it. Every time you power off the Snapmaker, you have to get screen authorization. Once connected to a specific device, it should be trusted so each new connect request, even after reboot of the printer, will automatically connect.

Eddie-Hartman, 2021-02-13:
With the latest firmware, I am able to connect and disconnect without confirming via the snapmaker touch screen every time.
Others have confirmed this on the forum as well. It was a somewhat hidden feature in the latest update.

christiangenco, 2021-03-26:
After I've initially connected (with screen authorization) I can reconnect without authorization, but I have to re-do this authorization any time power on the SnapMaker gets turned off.

I'd love if the "authorize" button remembered each computer I've authorized so I only ever had to authorize on the touch screen once per computer.

polrus, 2021-08-11:
printing is loud so if one has the possibility to print in another room/part of home it is done this way. Having to go to printer and back every time is pure annoyance

parachvte, 2022-12-22:
This feature should be supported in latest firmware (both SM 2.0 & J1 firmwares). You only need to authorize once the first time Luban connects to the printer, and later connections are auto-connected via authorized tokens.

Closing this issue.

closed 2022-12-22 by parachvte

poltpolt, 2022-12-22:
it works if you have static IP - if your printer  received another IP from your router you have to confirm again.

parachvte, 2022-12-22:
> if your printer received another IP from your router you have to confirm again.

Confirmed.

Luban identify printer by IP, it does not know another printer with different IP is the same printer or not. Only the printer with saved IP can be identified as the same printer. I know it's little silly, most of our users have only one SM printer at home, any printer discovered is that only printer 🥲 

I'll open a new issue addressing this. Guess we can have a quick fix for this, and for a long run, we need the firmware to have a unique identifier instead of IP.