Snapmaker/Luban #540: Cannot connect via USB to SM2

TurboTurtle, 2020-11-09:
I have the SM2 A350, using Luban 3.10.3 on Fedora 32, kernel `5.8.16-200.fc32.x86_64`

I cannot connect to the printer via USB. When I connect the printer to the system and refresh the serial port list in Luban, it detects /dev/ttyUSB0 properly, however when I click connect, I get a `Can not open this port` error.

`dmesg` shows the following after connecting the printer and a failed attempt to connect from Luban:

```
[643382.107714] usb 1-4: new full-speed USB device number 4 using xhci_hcd
[643382.337834] usb 1-4: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.63
[643382.337838] usb 1-4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[643382.337840] usb 1-4: Product: USB2.0-Serial
[643382.346590] ch341 1-4:1.0: ch341-uart converter detected
[643382.360611] usb 1-4: ch341-uart converter now attached to ttyUSB0
[643851.956341] show_signal: 77 callbacks suppressed
[643851.956343] traps: snapmaker-luban[404551] trap int3 ip:5620d2952cd3 sp:7ffe5eaf89c0 error:0 in snapmaker-luban[5620d0acf000+53d6000]
[643899.127395] traps: snapmaker-luban[404703] trap int3 ip:5627bffafcd3 sp:7ffd453ebe40 error:0 in snapmaker-luban[5627be12c000+53d6000]
```

jane-rose, 2020-11-16:
This is a problem with the fabscreen. Contact `support@snapmaker.com` and send a video to them. And technical support will contact you. Thanks

jane-rose, 2020-11-16:
I will close the issue.

TurboTurtle, 2020-11-16:
I don't believe that's accurate - I'm able to connect fine via Windows. The `Can not open open` message only appears when trying to connect on Fedora.

TurboTurtle, 2020-11-17:
@jane-rose So I was already getting a replacement screen from customer service. That's now arrived and installed, updated to latest firmware, and the issue still exists.

jane-rose, 2020-11-18:
> @jane-rose So I was already getting a replacement screen from customer service. That's now arrived and installed, updated to latest firmware, and the issue still exists.

ok, i got u

ldp7, 2020-11-18:
@TurboTurtle 
I cannot find any useful information with the username "TurboTurtle". Please contact support@snapmaker.com for further investigation on this issue.

Best regards

ldp7, 2020-11-18:
@TurboTurtle 

It seems that you have chosen the wrong port and the log was not useful at all. You can unplug the cable and check the right port, which will disappear in the COM port list. 

Please send me the email you used for the tickets so that I can follow the threads on your issues.

Best regards

jane-rose, 2020-12-29:
Since no further progress. I will close the issue.

ldp7, 2021-01-03:
For some time now I have been unable to connect to the SM2 printer via Luban over USB. I am running Fedora 32, Luban 3.12.3, and SELinux in either permissive or enforcing mode does not make a difference.

The controller is properly detected, as evidenced by `dmesg`:

[704825.527981] usb 1-4: new full-speed USB device number 4 using xhci_hcd
[704825.760568] usb 1-4: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.63
[704825.760570] usb 1-4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[704825.760571] usb 1-4: Product: USB2.0-Serial
[704825.768301] ch341 1-4:1.0: ch341-uart converter detected
[704825.782336] usb 1-4: ch341-uart converter now attached to ttyUSB0

Hi @jane-rose  Here is a issue about the Fedora 32, Luban 3.12.3. Please re-open this issue because you guys did not solve it.