Snapmaker/Luban #545: Erratic settings behaviors

ReperioFabrica, 2020-11-12:
Have not seen coding act like this since the 90s.

Project settings FREQUENTLY randomly ignored:

Adhesion - Example: Set to brim, does a skirt, or maybe none at all. Have to change settings back and forth and compile gcode until the correct 'adhesion'  setting "sticks."

Infill: I almost always do 100%, but Snapmaker Luban sometimes decides differently. One project in particular refuses to show and infill when set to 100% (the first time I printed this job it was set to 100 but printed about 15% which ruined the top). I have closed and opened the program and double checked the file and it refuses to show any infill when set to 100%.

Speed: Similar to the two aforementioned issues.

I finally for settings for various jobs and filaments dialed in, but getting the printer to listen is entirely ruining the flow. I don't mind wasting filament but the hours long print jobs need to be printed AS DIRECTED.

jane-rose, 2020-11-16:
Sorry for the bad experience, we will follow up and any progress we will catch u up.

sumibin, 2023-03-21:
This is an old obsession and we should have solved it.

closed 2023-03-21 by sumibin