Snapmaker/Luban #554: Luban wan’t quit correctly on Mac OS

Airmonkdev, 2020-11-13:
Hello,
I’ve installed Luban v3.10.3 on my MacBook Pro 13" 2020 but if I quit the application (CMD+Q, Quit from menu…) the app seems not really quit and even if the windows is closed the app remains on the dock. To quit the app I have to right click on the icon dock and select Quit or force quit from Apple menu.

Can you please fix it?

parachvte, 2020-11-14:
Another CMD+Q will quit the application.

Airmonkdev, 2020-11-14:
Yes it works but why we have to apply CMD+Q 2 times? Can you please fix it with the next release?

halr9000, 2020-11-14:
Agreed with OP, this violates the principle of least surprise on a mac.

ronzelver, 2020-11-17:
Also reboots are blocked by this behaviour, since MacOS cannot shutdown the application in a normal fashion.

kb1ibt, 2020-12-04:
This is a bigger problem than just the small section that @parachvte worked on in his merge. The answer in this StackOverflow explains what I am talking about. https://stackoverflow.com/questions/58822323/for-electron-app-in-macos-when-is-app-quit-invoked
In the File menu there is a New File option however that does not actually create a new workspace, the Quit menu doesn't actually close the program it only closes the current window. The Quit menu item should be calling the electron code "will-quit". The "window-all-closed" should not be closing the app, it should allow a new file/window to be created.

On top of all of these is the fact that the Preferences... from the Snapmaker Luban menu (or the Cmd+, keyboard shortcut ) does not open anything, my guess is it should be loading the Settings view.

parachvte, 2020-12-04:
@kb1ibt 

> In the File menu there is a New File option however that does not actually create a new workspace, the Quit menu doesn't actually close the program it only closes the current window. The Quit menu [...]

Let me make it clear, common practice is:

- Quit (`Cmd + Q`) should close all windows and quit the app. ("window-all-closed" is not emitted, "will-quit" is emitted)
- Close (`Cmd + W` is commonly used) should close current window. ("window-all-closed" may be emitted, but won't quit the app, "will-quit" is not emitted)
- New (`Cmd + N` or `Cmd + Shift + N` is commonly used) should be create a new window.

This was what I intended to work on. So there was code that when "window-all-closed" emitted but not quitting the app (macOS / darwin).

In the PR #595 it's reverting back to earlier version to support Quit (`Cmd + Q`) action. While Close (`Cmd + W`) action and New (`Cmd + N`) action are not implemented as window (It only refresh the Canvas). And this is what we should do next.

parachvte, 2020-12-04:
> On top of all of these is the fact that the Preferences... from the Snapmaker Luban menu (or the Cmd+, keyboard shortcut ) does not open anything, my guess is it should be loading the Settings view.

There was a binding `Cmd + ,` to open the Preferences page. But it's binded as global keyboard event, which means whatever app you are focusing on, `Cmd + ,` would open Luban's Preferences page. I removed the keyboard shortcut in previous version.