Snapmaker/Luban #564: Add Multi-Thread support

supra89t, 2020-11-16:
Hello!

I just finished installing Ubuntu Desktop 20.04.1 on my DELL PowerEdge R630 with 40 CPU threads and 128GB of RAM... and when I run the slicer its utilizing only 1 thread and its pinned at 100%. It will alternate with other threads but only 1 is pinned at a time. The others bounce around 10-20%, hard to tell if they are running for this application, and most are at 0%.

I tested the same model on another slicer program and it took 10x less time to slice. While monitoring the CPU usage its visible that for a portion of the task it does indeed use all 40 threads. When creating the tool paths it drops down to 1 thread. However the majority of the tasks are multi-threaded.

This addition will not only help me but most other users as well, seeing as most computers have multi-threaded CPU's. 

I'm willing to test, provide reports or even screenshots if needed.

Thank you.

mr-manuel, 2020-11-16:
I see the same behavior on Windows 10 x64 with Luban 3.11.0, that only 1 core is used from Luban. I would appreciate it, when the whole power can be used

rbcapp, 2021-01-03:
Partial investigation notes:
Cura engine called from this function: https://github.com/Snapmaker/Luban/blob/191b367f6e17ab11206595ee1cdc7d3da3482d78/src/server/slicer/slice.js#L47
Cura supports thread count setting by command line via "-m<thread_count>" if OPENMP is defined.

Still need to research: 
- What are the compilation settings for CuraEngine.exe?
- Would this change the installation needs for Luban?
- Thread availability determination in Luban slice.js#L47

sumibin, 2023-03-21:
Multithreading is now supported in both the Slicer and Toolpath engines. However, it should be noted that some slicing processes still use single threads, which need to be solved by algorithmic optimization.

closed 2023-03-21 by sumibin