Snapmaker/Luban #569: Prepare 3D printer for quick start

JohnAllijn, 2020-11-19:
If I start the printer there are a number of actions that I take to get ready to print. They are always the same and take up about 5 minutes:

1) turn on the printer
2) preheat nozzle
3) preheat printer bed
4) turn on LED lights in the enclosure
5) connect from my Mac to the printer over wifi
6) accept the remote connection on the printer

and then I am ready to print. 
Can you please make a preset of those actions? Maybe a button that I can press on the controlscreen of the printer that does all that while I download the model to the printer?
thanks,
John Allijn

parachvte, 2020-11-20:
Oh there is a Macro widget in "Workspace", you can use it to group G-code commands together. While some of these commands are HMI (Touch Screen) commands (via HTTP API), we don't support them in G-code.

Also heat the nozzle and heated bed are in G-code Start codes, you don't really need to preheat them.

So,
1. manually
2. in G-code
3. in G-code
4. HTTP API (currently only manually in Luban or HMI)
5. HTTP API
6. A bit annoying the do the confirmation every time, we will optimize the process that you will only need to confirm once.

mr-manuel, 2020-11-20:
For 6. there is already a feature request open: [Feature Request: allow automatic wi-fi connection #535](https://github.com/Snapmaker/Luban/issues/535)