Snapmaker/Luban #578: "Unable to start the job" error during CNC operation

halr9000, 2020-11-24:
While doing some basic CNC operations, running boundary, clicking play to run job, I have seen this error twice and no clue how to troubleshoot. Contrary to appearances, it’s not fatal. My CNC job is running right now behind the error message, and before that the boundary run completed successfully. Clicking “X” dismisses the error and I can go on my way. Just seems like some bad error control at the very least.

OS: Win10
Luban version: 3.11.0
Connecting over wifi
Printer: SM2 A350 (latest firmware but can't get version without disconnecting wifi which is a pain)

[image]

Cross-post to forum https://forum.snapmaker.com/t/unable-to-start-the-job-error-during-cnc-operation/11807

mpuzar, 2020-11-27:
I can confirm the problem occurring, sometimes even while not doing anything directly with the machine and the machine being idle. I don't see whether this error affects anything or whether it is only a false alarm and thus just being a nuisance.
OS: Win10
Luban version: 3.10.2
Connecting over wifi
Printer: SM2 A250 (latest fw as well)

jane-rose, 2020-11-30:
Sorry for that happened.
The wifi connection of version 3.10.2 is a bit unstable. Version 3.12.0 will be released this week with optimized wifi connection, will you try again when it is released later?

parachvte, 2020-12-01:
Can you check if the machine is in PAUSED or PRINTING state?

The problem could occur when the machine is NOT in IDLE state, or not finish previous job properly.

closed 2022-12-22 by parachvte

jalapenopuzzle, 2025-03-23:
I saw this issue. Noting the circumstances here for future google reference.

I ran a CNC job using a GCode file generated from FreeCAD, using an updated post-processor script.
My initial test of the CNC job resulted in the bit stalling and the machine stopped the job.
Subsequent attempts to start a new job (with altered GCode to address the reason for the bit stall) resulted in the following message on the touch screen:

```
Unable to start the job.
Please check your machine settings
```

(Message may not be exactly correct.)

I am still able to move the toolhead using the controls, so the emergency stop function has not been activated.

Restarting the machine allowed me to process the next job without seeing this error.

Machine: Snapmaker 2.0 A350T
Firmware: Snapmaker2_V1.20.3_20240716

FreeCAD: 1.0.0
post-processing script: hoping to publish it soon subject to testing