Snapmaker/Luban #590: Raspberry Pi Support

soberwalder, 2020-11-27:
I tried to compile Luban for my Raspberry Pi 4 but I failed somewhere in the long list of dependencies.
It would be great if you can provide a binary package for Raspberry.

parachvte, 2020-12-01:
Which distro of Pi are you using, Raspbian? Not sure if the .deb version can be used.

soberwalder, 2020-12-01:
```
pi@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

I'm afraid I can't use the .deb version because it is compiled for x86 and the Raspberry has an ARM processor. But I have not tried it either.

soberwalder, 2020-12-01:
Just tested the install of the .deb file which failed as expected.

deimi, 2021-01-27:
@soberwalder have you also tried to use Ubuntu 20.04 on the raspberry instead of Raspbian?

soberwalder, 2021-01-27:
Hi @deimi 
no, I've not yet tried to install on Ubuntu but it will not install because of the different CPU architecture of the Raspberry Pi (arm64). The current deb file has amd64 architecture.

deimi, 2021-01-27:
yes, I'm aware of this, so I was not talking about the deb file but maybe the compilation works there

soberwalder, 2021-02-01:
Took me a few days - needed to order a new SD card for testing this

Installed Ubuntu 20.10
> lsb_release -a
> No LSB modules are available.
> Distributor ID:	Ubuntu
> Description:	Ubuntu 20.10
> Release:	20.10
> Codename:	groovy

Installed git, libgconf, binutils, rpm and nvm
Cloned the Luban repo

npm install gave me following ouput
> npm install

> @serialport/bindings@8.0.8 install /home/stefan/prog/Luban/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

> prebuild-install WARN install No prebuilt binaries found (target=10.23.2 runtime=node arch=arm64 libc= platform=linux)
> gyp ERR! build error 
> gyp ERR! stack Error: not found: make
> gyp ERR! stack     at getNotFoundError (/home/stefan/.nvm/versions/node/v10.23.2/lib/node_modules/npm/node_modules/which/which.js:13:12)
> gyp ERR! stack     at F (/home/stefan/.nvm/versions/node/v10.23.2/lib/node_modules/npm/node_modules/which/which.js:68:19)
> gyp ERR! stack     at E (/home/stefan/.nvm/versions/node/v10.23.2/lib/node_modules/npm/node_modules/which/which.js:80:29)
> gyp ERR! stack     at /home/stefan/.nvm/versions/node/v10.23.2/lib/node_modules/npm/node_modules/which/which.js:89:16
> gyp ERR! stack     at /home/stefan/.nvm/versions/node/v10.23.2/lib/node_modules/npm/node_modules/isexe/index.js:42:5
> gyp ERR! stack     at /home/stefan/.nvm/versions/node/v10.23.2/lib/node_modules/npm/node_modules/isexe/mode.js:8:5
> gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:153:21)
> gyp ERR! System Linux 5.8.0-1006-raspi
> gyp ERR! command "/home/stefan/.nvm/versions/node/v10.23.2/bin/node" "/home/stefan/.nvm/versions/node/v10.23.2/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
> gyp ERR! cwd /home/stefan/prog/Luban/node_modules/@serialport/bindings
> gyp ERR! node -v v10.23.2
> gyp ERR! node-gyp -v v5.1.0
> gyp ERR! not ok 
> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.3 (node_modules/watchpack/node_modules/fsevents):
> npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})
> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/fsevents):
> npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})

> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! @serialport/bindings@8.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
> npm ERR! Exit status 1
> npm ERR! 
> npm ERR! Failed at the @serialport/bindings@8.0.8 install script.
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

> npm ERR! A complete log of this run can be found in:
> npm ERR!     /home/stefan/.npm/_logs/2021-02-01T20_36_05_852Z-debug.log

Logfile attached
[2021-02-01T20_36_05_852Z-debug.log](https://github.com/Snapmaker/Luban/files/5906692/2021-02-01T20_36_05_852Z-debug.log)

Same error as on Raspbian by the way.

Tried npm run dev with the following output

> npm run dev

> snapmaker-luban@3.13.0 dev /home/stefan/prog/Luban
> npm run build-dev && npm run start-dev

> snapmaker-luban@3.13.0 prebuild-dev /home/stefan/prog/Luban
> bash build/prebuild-dev.sh

> snapmaker-luban@3.13.0 pkgsync /home/stefan/prog/Luban
> node build/pkgsync.js

> ~/prog/Luban/src ~/prog/Luban
[... 72 lines cut ...]
> [21:39:00] [nodemon] watching extensions: js,json
> [21:39:00] [nodemon] bind restart -> `osascript -e 'display notification "App restarted due to:
> '$FILENAME'" with title "nodemon"'`
> [21:39:00] [nodemon] starting `node ./bin/cli`
> [21:39:00] [nodemon] forking
> [21:39:00] [nodemon] child pid: 50286
> [21:39:00] [nodemon] watching 188 files
> Error:  Error: There is no built binary for font-manager
>     at Object.<anonymous> (/home/stefan/prog/Luban/node_modules/font-scanner/index.js:8:11)
>     at Module._compile (internal/modules/cjs/loader.js:778:30)
>     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
>     at Module.load (internal/modules/cjs/loader.js:653:32)
>     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
>     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
>     at Module.require (internal/modules/cjs/loader.js:692:17)
>     at require (internal/modules/cjs/helpers.js:25:18)
>     at eval (webpack-internal:///font-scanner:1:18)
>     at Object.font-scanner (/home/stefan/prog/Luban/output/server/index.js:4505:1)
>     at __webpack_require__ (/home/stefan/prog/Luban/output/server/index.js:21:30)
>    at eval (webpack-internal:///./lib/FontManager.js:23:71)
>     at Module../lib/FontManager.js (/home/stefan/prog/Luban/output/server/index.js:3118:1)
>     at __webpack_require__ (/home/stefan/prog/Luban/output/server/index.js:21:30)
>     at eval (webpack-internal:///./lib/svg-convert.js:18:70)
>     at Module../lib/svg-convert.js (/home/stefan/prog/Luban/output/server/index.js:3750:1)
> [21:39:03] [nodemon] clean exit - waiting for changes before restart

parachvte, 2021-02-02:
serialport (https://github.com/serialport/node-serialport) is not properly installed.

In short:
1) it doesn't have pre-built binary for (target=10.23.2 runtime=node arch=arm64 libc= platform=linux) 
2) and you don't have `make` installed to build on your local machine.

for 1), we need @serialport/bindings pre-built binary. I've searched [@serialport/bindings@8.0.8](https://github.com/serialport/node-serialport/releases/tag/%40serialport%2Fbindings%408.0.8), 
The version of Electron we used is v76, but for some reason the binary is missing, only version up to `bindings-v8.0.8-electron-v75-linux-x64.tar.gz`.

To resolve this, either **downgrade Electron version** or **upgrade serialport to v9**.
**upgrade serialport to v9** is much simpler, you can start from there.

soberwalder, 2021-02-02:
1. I've installed make and build-essentials (thx for the hint)

2. `npm install serialport`

>  @serialport/bindings@9.0.4 install /home/stefan/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
> ...
> + serialport@9.0.6
> 
> added 64 packages from 43 contributors and audited 78 packages in 19.978s
> 
> 5 packages are looking for funding
>   run `npm fund` for details
> 
> found 0 vulnerabilities

3. `npm install` 
> snapmaker-luban@3.13.0 prepare /home/stefan/prog/Luban
> npm run clean

> snapmaker-luban@3.13.0 clean /home/stefan/prog/Luban
> bash -c 'rm -rf ./dist ./output'

> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.3.1 (node_modules/stylint/node_modules/fsevents):
> npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.1: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})

> audited 2491 packages in 44.493s

> 81 packages are looking for funding
>   run `npm fund` for details

> found 11 vulnerabilities (4 low, 5 moderate, 2 high)
>   run `npm audit fix` to fix them, or `npm audit` for details

4. `npm run dev`
> ...
> Error:  Error: There is no built binary for font-manager
>     at Object.<anonymous> (/home/stefan/prog/Luban/node_modules/font-scanner/index.js:8:11)
>    at Module._compile [...]

Installed font-manager with `sudo apt-get install font-manager` but it didn't change anything.

Tried `npm install font-manager` with a few warnings but the `npm run dev` still failed with the "There is no built binary for font-manager"

soberwalder, 2021-02-02:
I cloned the git repo in a new folder and it suddenly compiled.
Going to test it with a clean install of Ubuntu again, write down the steps and try it under Raspian OS afterwards.

ITmaze, 2021-02-03:
When you describe the steps, please share them here. I've been working on a
Dockerfile to make stand-alone containers with Luban installed and using
the .deb is giving me missing dependencies. I haven't yet managed to
determine what they should be, so I haven't yet created an issue to report
it, but it seems related to your experience.

o

soberwalder, 2021-02-03:
@ITmaze 
I've setup a fresh installation of Ubuntu 20.10 and installed the following packages before the Development section in the README.md
`sudo apt-get install git make build-essential libfontconfig-dev curl`

I did the same on my Raspbian OS installation and it worked too.

@parachvte 
It looks like it worked with serialport 9.0.4. Do you want me to change this in packages.json?
Besides this change I'd add libfontconfig-dev as additional package for the development environment in the README.md. It's needed to build the font-manager. 
After that it should be possible to provide binary packages for Raspberry Pi

parachvte, 2021-02-04:
@soberwalder 
Yes please open a pull request merging branch `master`. We'll test it too on macOS and Windows.

If we use Travis-CI to build executable, probably we need to add configuration for Electron to build ARM version.
Or do you think it is ok to add document instructions on compilation for Raspberry Pi / Dockerfile?

ITmaze, 2021-02-04:
I'm in two minds.

I've now attempted to install using the repo, the tgz and the deb inside Docker using a variety of options and dependencies. Normally I'd say that if there is a "standard" linux installation/build instructions, a Docker installation can pretty much follow the same path.

On the other hand, there are several different things you can build using Docker. There is the ability to have a Docker image represent a running instance of snapmaker-luban, complete with GUI, as-if it's running as a locally installed application. Another option is to use Docker to create the installation packages, the tgz, the deb, the exe, the rpm etc. This could be used to generate all the build artefacts. Finally, Docker could be used to run snapmaker-luban as a web-server that can be accessible from a browser.

I think we need to first update the README.md to have a complete set of instructions. For example, nvm isn't a standard package in any of the linux images I have seen and installing it seems non-trivial.

How about splitting README.md into two, with a DEVELOPMENT.md discussing the build process, and leaving the README.md with end-user artefact installation instructions. Installing nvm or build-essentials isn't something that an end-user would typically do.

Finally, I'd advocate for including a Docker based build process. I've started the process of making something that can build GUI app containers for each release, but that's only part of the puzzle.

Thoughts?

jane-rose, 2021-02-04:
A few months ago, I tried to do it on Raspberry 3 b+. I don't know if it helps. Maybe you can try this. https://github.com/jane-rose/OctoLuban

tyeth, 2022-03-05:
Did you progress on this? I might just run it on a Windows pc and expose it to the network, but putting on the pi makes so much more sense. Also integrating with octoprint as a slicer style plugin would be awesome. I used plugins in the past running slicer and cura inside octoprint quite happily. It would need a tweak as to whether to have an extra queue in octoprint option as well as the usual export gcode / load in workspace

alexastro41, 2025-03-05:
I have tried a few different things. I have even got it "running" on the raspberry pi 4 and have run into it hanging and crashing doing normal operations. My latest approach after reading some electron blurbs was to cross compile on x64 linux using a VM for arm64 for the pi. So far I can cross compile (by modifying the packages slightly) and install on the raspberry pi. The problem is that it seems the font-manager cannot be found by the font-scanner package. I am unsure what this fix might entail as it could be due to this project, font-scanner or the latest version of Raspberry Pi OS.