Snapmaker/Luban #593: Issue saving laser process parameters

Oachkatzale, 2020-11-29:
unfortunately the laser process parameters are not saved when saving the project. Finding the right process parameters is one of the biggest challenges for most laser projects. It would be very useful if the parameters could be saved with the projects 
 (SNAPLZR).  This could save a lot of work and would 
avoid stupid mistakes.
[image]
[image]

jane-rose, 2020-11-30:
Sorry, it in the schedule.  Any progress will update in this issue. Thanks! #534

snoeberer, 2020-12-28:
hello parachvte, do you have an idea in which version this bug will be fixed? maybe an rough schedule?

parachvte, 2021-04-27:
With v3.15.0, laser parameters should be saved in project.

closed 2021-04-27 by parachvte