Snapmaker/Luban #614: svg parser does not respect hidden layers

rnoah, 2020-12-08:
Some tools such as Inkscape have the ability to hide elements or layers by setting the property _style="display:none"_ as in the following snippet generated from Inkscape:

  '<'g
     style="display:none"
     inkscape:label="Layer 2"
     id="layer2"
     inkscape:groupmode="layer">'
    <<path
...
Luban (laser module) renders the elements from the svg file that should be hidden.

For reference from: https://developer.mozilla.org/en-US/docs/Web/SVG/Attribute/display

A value of display="none" indicates that the given element and its children will not be rendered. Any value other than none or inherit indicates that the given element will be rendered by the browser.

jane-rose, 2020-12-08:
Sorry for that. At this moment , we are not able to parse `class` and `style` in SVG parser.

TriploidTree, 2023-12-30:
I'd like to bump this feature. It's really useful to have hidden layers on an SVG and having to keep a second one for export is a bit frustrating.

Michael-Longley, 2024-06-03:
New to the repo, so this may be incorrect, but I see styles being parsed in [SVGLoader.js](https://github.com/Snapmaker/Luban/blob/main/src/app/scene/three-extensions/SVGLoader.js#L838). A quick look at the function makes me think you should be able to add something like 
```
if (node.hasAttribute('style')) {
    styles = node.getAttribute('style').split(';')
    for (let i = 0; i < styles.length; i++) {
        stylesheetStyles = Object.assign(stylesheetStyles, styles[i])
    }
}
```

I haven't done a deep dive into the code, so I may be wrong here but at a first glance that seems correct. In addition to that, the `style="display:none"` works in the preview it only seems to fail in the GCode generation, as such, this is a solved problem elsewhere in the code.

Unfortunately I don't have a copy of the repository with build running locally so I'm unable to test. I would also greatly appreciate this feature though. My current work flow is Export image from InkScape, open SVG in VSCode, CTRL-F for "display:none", delete associated nodes, repeat.