Snapmaker/Luban #617: M1 native binary for macOS

rstocks, 2020-12-09:
Please consider compiling with M1 / Arm processor support for macOS.

parachvte, 2020-12-10:
I haven't read through the issue (https://github.com/electron-userland/electron-builder/issues/5392) about macOS/ARM64.
And we don't have a M1 in hand...

If we add ARM64 build, could you test the package for us?

rstocks, 2020-12-10:
> If we add ARM64 build, could you test the package for us?

Totally! Point me at 'em :-)

SketchWork, 2021-09-13:
Same here.  Please point me at a testing version for Apple Silicon (M1).

daiku1, 2021-12-10:
Also would test Apple m1 processor.  Point me to the testing version

aminutolo, 2022-12-04:
I know this request is a little older, but I am also happy to test or contribute if provides any benefit. I would also like to see a native build for the ARM64.

parachvte, 2022-12-07:
Please checkout: https://github.com/parachvte/Luban/releases/tag/v4.5.0-beta.4

There is a test build for ARM64 architecture, [Snapmaker.Luban-4.5.0-beta.4-arm64.testing.dmg](https://github.com/parachvte/Luban/releases/download/v4.5.0-beta.4/Snapmaker.Luban-4.5.0-beta.4-arm64.testing.dmg).

@daiku1 @aminutolo @SketchWork @rstocks

closed 2023-01-16 by parachvte