Snapmaker/Luban #618: Feature request: Integration with Octoprint

rstocks, 2020-12-09:
While the wifi capabilities of the Snapmaker are great, and something I have used (especially with the laser module), for various reasons I prefer to manage my 3D print jobs with Octoprint. Adding direct integration with the Octoprint API (manage gcode files on the Octoprint server, control jobs, etc) would be great.

Octoprint API details: https://docs.octoprint.org/en/master/api/index.html

parachvte, 2020-12-10:
It's relatively low priority in our backlog (for a while). The request was brought up in the forum before.
We don't have enough crews to handle OctoPrint support for now. Thanks for understanding.

rstocks, 2020-12-10:
Low priority is totally fine; I just wanted to have it written down somewhere official :-)