Snapmaker/Luban #632: [BUG] Wrong scaling of DXF drawings version issue from 3.9 - 3.12.3

TheTravel007, 2020-12-13:
Dear Lubanteam, 

Within the latest version of Snapmaker Luban the import of dxf drawings does not work correctly anymore.
The dimension of the drawings are wrong. I attached a few screenshots of a simple rectangle dxf drawing. 

The original drawing size is 63 mm x 6 mm. 

[image]
[image]

Once it is imported to Luban it changes it to 8.3 x 1 mm. 

[image]

This scaling issue seems to be a problem since version 3.9. 
In 3.8 the scaling is correct, it imports the drawing without any problems. 

Please see attached the dxf drawing for your reference. 
[Bar.zip](https://github.com/Snapmaker/Luban/files/5684833/Bar.zip)

rkreienbuehl, 2020-12-14:
I have encountered the same issue by loading SVG's for CNC or laser engraving.

jane-rose, 2020-12-14:
Sorry for that, same issue with #502. We may fix it after one version. At this time,u can change the size manually.

WendyLud, 2021-01-27:
Laser function 3.13.1: Testing importing a dxf with a rectangle 200 x 100 mm shows up as a mini rectangle about 13,1 x 25,9 mm . Correcting the size via Width and Height to 200 x 100 mm show the handles become 200 x 100 mm in Edit mode. The actual rectangle is still smaller.

parachvte, 2021-01-27:
The size divided by a constant scale value (7.8).

We can confirm that the issue is still existing in version 3.9.x to 3.13.x. And it should be fixed in version v3.15.0 (in March).

WendyLud, 2021-01-27:
I hope this can be fixed earlier because this makes the laser useless for me.

DIYgitylDreams, 2021-01-28:
when inputing the correct dimensions into luban, the dxf values are still wrong.  Luban seems to be adding a buffer around the parameter of the image.  Any idea on the value that needs to be added to compensate for the buffer?

achimb, 2021-01-31:
Hi there! I encounter the same problem. I check the source code (but I can't run it in a debugger thought) and come to the conclusion that the problem is coming from this:

In ThreeDxfLoader.js, Line 738 - 743 in function load(path, onLoad) a "scale" variable is initialised with 1 (which makes totally sense to me):

```
            let scale = 1;
            this.measureBoundary(dxf);
            if (this.width) {
                scale = this.width / dxf.width;
            }
            this.normalizer(dxf, scale);
```

If the "this.width" member variable is not nil then the "scale" variable will be changed from 1 to something else (in my opinion unpredictable). Further down the road: The "this.width" member variable gets loaded in the construct of the class itself in line 82:

```
class ThreeDxfLoader {
    constructor(options = {}) {
        const { font, width, height } = options;
        this.font = font;
>>>>>>  this.width = width;
        this.height = height;
    }
```

The "width" parameter gets filled in when loading the dxf file in the source file Model.js in lie 163: 

```
 generateModelObject3D() {
        if (this.sourceType === 'dxf') {
            if (this.modelObject3D) {
                this.meshObject.remove(this.modelObject3D);
                this.modelObject3D = null;
            }

            const path = `${DATA_PREFIX}/${this.uploadName}`;
>>>>>>    new ThreeDxfLoader({ width: this.transformation.width }).load(path, (group) => {
                this.modelObject3D = group;
                this.meshObject.add(this.modelObject3D);
                this.meshObject.dispatchEvent(EVENTS.UPDATE);
            });
```

This function in turn fills in the this.transformation.width parameter which seems a bit strange, because at this point the width of the dxf object isn't known yet.

Unfortunately I can't debug the running code so by just reading the code I am stuck here. I am not sure where the this.transformation.width would come from. I think the constructor for ThreeDxfLoader() shouldn't get a "width" with it, so it is still nil when ThreeDxfLoader.load() is checking on it and if the width is then nil the scale won't be altered from 1.0 in the first place. However, my guess is that if this all is changed to be nil, the user won't be able to change the dimensions anymore at all?!

BTW: @DIYgitylDreams If you want to put in the correct dimensions in the UI for your DXF object, you should put in 2 mm more for the width (or height) than your original dxf export file as this code snippet suggests: (Luban is adding 1mm around your design, I guess for UI purposes)

ThreeDxfLoader.js, line 721-726:

```
    measureBoundary(dxf) {
...
        dxf.boundary = {
            minX: minX - 1,
            maxX: maxX + 1,
            minY: minY - 1,
            maxY: maxY + 1
        };
```

I hope this analysis will help to solve the issue ASP! Thanks!

ITmaze, 2021-02-02:
To debug running Luban code, you can press Ctrl-Shift-C and enable the
chrome console. There is a debugger built-in.

theozhbl, 2021-11-24:
Same issue as @ozzyman with Luban V.4.0.3
Is there any progress or a workaround on this?
I remember similar cases where the scaling was wrong by a constant factor of 25.4 (guess why?), but here it seems to be some weird unknown (random?) number. Even if I scale my e.g. 40x40 mm page size SVG from Inkscape in Luban to 40x40 mm the output to PCB will be somewhat smaller. So, I have to scale my SVG to something like ca. 41x41 mm to get the carved holes and lines at the right position. How to find this number? Any ideas to workaround this very annoying bug?

achimb, 2021-11-24:
@theozhbl  As I mentioned in my post from January, as a workaround the size needs to be 2 mm bigger than the original boundaries:

"If you want to put in the correct dimensions in the UI for your DXF object, you should put in 2 mm more for the width (or height) than your original dxf export file as this code snippet suggests: (Luban is adding 1mm around your design, I guess for UI purposes)"

I hope this works for you!

closed 2023-01-18 by parachvte

Voodoofly, 2023-12-11:
Hi Guys,  today in year 2023  still the same BUG !!! 
Luban is up to date ! 

i export from Fusion360  and the objekt in luban is so 10x smaller then it must be  !!! ?  

Why cant fix this after 3 Years ?
THX

CHristapherenese, 2023-12-20:
> Hi Guys, today in year 2023 still the same BUG !!! Luban is up to date !
> 
> i export from Fusion360 and the objekt in luban is so 10x smaller then it must be !!! ?
> 
> Why cant fix this after 3 Years ? THX

It just started again for me on the last update 
[image]