Snapmaker/Luban #635: Luban hangs and fails to create G-Code when there are multiple bodies in the build area

jgbenav, 2020-12-14:
Version 3.12.3
Machine A350
64 Bit
Windows 10 Pro 64bit
RAM 116 GB
Intel Core i7 @2.8 ghz

When loading multiple parts into the build area, Luban performance begins to degrade as soon as the third part is added.  When I get over three parts (even if they are the same parts added over and over again) Luban will just hang after pressing the Generate G-Code button.  At first it will say preparing to slice and sit there for a while and eventually the whole window turns white except for the top line Menus.  The menus are still clickable and function.  The only way to get the screen back to normal is to click on "VIEW" and select "RELOAD"
The parts that I was making at the time were small very simple prismatic parts.  No complexity at all.  I did an experiment and added 9 parts that were a little more complex and the same issue occurred.   I took those sample parts and added a bridge between them to join them all into a single body then imported them into Luban and sliced it.  It worked fine.   So part size and complexity do not seem to be a factor.  Only multiple solid bodies that are disjoint give me this problem

isaacrlevin, 2021-02-03:
Seeing this as well, for instance thus STL file scaled to 70% hangs up 27.6 slicing 

https://www.thingiverse.com/thing:4711071/files

parachvte, 2021-02-03:
https://github.com/Snapmaker/Luban/pull/737 is trying to improve on this. Expected to be released on version 3.15.0.

closed 2023-03-21 by sumibin