Snapmaker/Luban #640: [BUG] Luban blanks out on connecting via wifi

rkreienbuehl, 2020-12-16:
When I try to connect Luban via wifi to my Snapmaker A250, first of all, it almost never finds the printer by sarching, I have to enter the IP address manually. When I hit connect, the normal screen appears with the message to confirm on the printer. As soon as I confirm the connection, Luban blanks completely out, meaning, I only have a white screen and I have to close the window and reopen it.
I am using Luban on a MacBook Pro with macOS 10.15.5
As I saw in the forum, this bug also is present on Windows: [forum post](https://forum.snapmaker.com/t/luban-3-12-3-problems/12667/5?u=rkreienbuehl)

tlaurion, 2020-12-21:
Same on Debian-10.

Wynand-S, 2020-12-25:
Same on Windows 10 x64 with Luban v3.12.3 as well as Luban 3.13.0
Bug occurs when connecting via IP to Snapmaker. The window blanks to white, and the Snapmaker actually gives me an option to 'disconnect'.
When I revert to Luban 3.11 it works once more.

tkessels, 2020-12-30:
I've got the same issue.

jane-rose, 2020-12-31:
> I've got the same issue.

What version of the software you used before, and does the bug only appear once or every time you connect wifi via ip

ReaHe, 2020-12-31:
I also have this issue, I had to downgrade to 3.11 to get it to work.

@jane-rose I got this bug every single time as long as I did a direct-connect. Did not test via other means as my desktop is not near the printer.

jane-rose, 2020-12-31:
> I also have this issue, I had to downgrade to 3.11 to get it to work.
> 
> @jane-rose I got this bug every single time as long as I did a direct-connect. Did not test via other means as my desktop is not near the printer.
> 
> Pre-release also had this issue for me still (3.13)

OK, at this time, we cannot reproduce this phenomenon. Can we think that 3.11.0 is available for connection, and when you upgrade from 3.11.0 to 3.12.3, it will not be available?

ReaHe, 2020-12-31:
@jane-rose upgrading to anything 3.11 causes issues, to possibly single it out my snapmaker is in a completely different then my desktop. So that is also why I'm direct connecting the first place. This may help reproduce the issue, as I believe none of us are to able to see the device via network broadcasts.

jane-rose, 2020-12-31:
which platform u are using? For macOs,

ReaHe, 2020-12-31:
@jane-rose Windows 10 64bit

jane-rose, 2020-12-31:
ok, can i get u email to send 

> @jane-rose upgrading to anything past 3.11 causes issues, to possibly single the specific problem, my snapmaker is in a completely different subnet then my desktop. So that is also why I'm direct connecting in the first place. This may help reproduce the issue, as I believe none of us are to able to see the device via network broadcasts.
> 
> My next guess is that I'm also nat'ing to the snapmaker, so the snapmaker is seeing the routers ip and not mine and somehow is causing issues.

At present, nat mode will casue the device not showing on th broadcasts, maybe u can change it to use bridge mode

ReaHe, 2020-12-31:
@jane-rose 
It would not be easy for me to do this as my network is setup to perform network monitoring this way and would require quite a bit of rework to use bridge mode. I do understand that this prevents the broadcasts, but I'm surprised this effects the direct connect when used with the ip.

achampion, 2021-01-02:
I can echo, this is also happening on my system - on manual connect. Mac OS Catalina. I haven't tried downgrading to 3.11 yet.

DaLucidLight, 2021-01-05:
I am also seeing this with my A350. I have the latest firmware and Snapmaker Luban 3.12.3. My computer is hardwired to ethernet, printer is on wifi. Printer is not detected, and when I connect manually to it's IP, Luban goes blank. I have windows 10 64-bit.

jane-rose, 2021-01-07:
@ReaHe  Sorry for the late, guys. Finnally we reproduce the bug, and later i will send the pr to fix it. Happy new year!

jane-rose, 2021-01-21:
#714

jane-rose, 2021-01-21:
Hi, everybody! A new version has been released and the bug was fixed. I will close the issue.

closed 2021-01-21 by jane-rose

duppypro, 2022-05-28:
I am seeing the same white screen when pressing Connect button after I update to Luban 4.3.0 today.

After restating the app and the printer. I no longer get a white screen but I do immediately get a "Failed to connect, Invalid authentication."

Luban Workspace is finding the printer and the IP shown in Luban matches the IP shown on the printer About menu.

Very frustrating to not be able to print until this is fixed.