Snapmaker/Luban #664: Pan, zoom, orbit controls

schulzch, 2020-12-27:
Luban uses mouse bindings which are pretty much unknown to me. This is an issue when switching applications. There are a couple of other mouse models and it would be really to have a setting for this. I've build a non-exhaustive list of mouse models for reference:

## Luban v3.12.3

  Pan: right mouse button
  Zoom:  mouse wheel
  Rotate: left mouse button

## Prusa Slicer v2.3:

Almost the same, except that the middle mouse button pans as well.

  Pan: right mouse button or middle mouse button
  Zoom: mouse wheel
  Rotate: left mouse button
  
## Tinkercad (also used by Chitubox and configurable in Fusion 360):

Left and right are swapped (this is what drives me crazy).

  Pan: left mouse or middle mouse
  Zoom: mouse wheel
  Rotate: right mouse

## Fusion 360

Has support for several models. See [Fusion 360 preferences](https://www.autodesk.com/products/fusion-360/blog/fusion-360-zoom-pan-orbit-preferences/).

## FreeCAD

Has support for several models. See [FreeCAD mouse models](https://wiki.freecadweb.org/Mouse_Model).

## Blender

Freely configurable. See [3D View navigation](https://docs.blender.org/manual/en/latest/editors/3dview/navigate/navigation.html).

edward-crouch, 2020-12-28:
It would also be really useful if Luban supported 3DConnexion devices like my Spacemouse Pro. I can make Luban zoom in and out from the spacemouse, and I've tried to map the other axes but no joy. I would imagine if there is an API covering this, it would "just" (HA! just......) be a job of making the function calls?

Ed.

schulzch, 2020-12-29:
> It would also be really useful if Luban supported 3DConnexion devices like my Spacemouse Pro. I can make Luban zoom in and out from the spacemouse, and I've tried to map the other axes but no joy. I [...]

Yes, adding support for those is pretty easy - even without their SDK. There seems to be a package for it https://github.com/microdee/hid.spacemouse . I gave mine away a few years ago, though we still have one at the institute I could access for testing.

kpninja12, 2021-01-31:
Seems the best option here would be to add some more advanced settings in the Settings menu where the user can custom map controls for each function (pan, orbit, zoom)

schulzch, 2021-01-31:
Yes, I think so too. Probably mapping input to the (generic) camera, along with some tweaks to the overall navigation (that negative-scaled matrix when zooming that results in flipping the entire coordinate system is quite annoying).

I was super busy with paper writing until Friday and had planned to look into it next weekend.

parachvte, 2021-01-31:
What we did is to align orbit control with popular CAD/CAM software as possible.
Though the control is not perfectly implemented, custom mouse bindings are beyond our plans recently.

But...PRs are welcome 😃

Woodie77, 2022-05-02:
Hi, 
has there been any activity for implementing the spacemouse in luban?