Snapmaker/Luban #673: Laser doesn't stop on Button Stopp or Pause

Pertrit, 2020-12-29:
The Buttons "STOPP" and "PAUSE" should make the machine instantly Stop its task without any Questioning or whatever. This does not work in Luban or on the Snapmaker Touchscreen itself.
I tried to use these button and the machine goes on for the full round before it stops. The problem is that had leveled the 3D-Printing-Module to low and the funktion didn't work thats why i have a big scratch in my board.

I hope that the issue will be fixed soon.
thanks, Pertrit

jane-rose, 2021-01-06:
Sorry for that happened. This may not fixed throught Luban or Touchscreen . But as far as I know, the emergency stop button will be launched soon.

Pertrit, 2021-01-06:
Yeah, the emergency stop is launched soon, but what the emergency stop ist doing should the software do too, I think. Cause if you calibrate the Machine or something it is extremely helpful if you know the machine is stopping if you want. And this at the time you click the stop button, not after completing the command. If this isn't possible in the software or on the touchscreen than the emergancy can only produce a power loss to stop the machine, This i can do too with a button or a switch in the power connection to the plug. But i dont want to shut down the machine, I only want that it stops. I think and hope thats possible.

closed 2023-03-21 by sumibin