Snapmaker/Luban #674: [BUG] Disappeared print settings

chrisywong, 2020-12-29:
After an update (I think it was to 3.12.3), print settings no longer show up on 3D Printing window (the side headers are expanded, not minimized). Reinstall/downgrading to 3.10.2/upgrading to 3.13.0 didn't help.  Also I can't load more than one model into the workspace, and it hangs on "Loading model" on the second model. Not sure if they're related. 
This is exactly like https://github.com/Snapmaker/Luban/issues/522 and https://forum.snapmaker.com/t/no-printer-settings-in-3-10/10940

[image]

Console right after startup:
```
THREE.WebGLRenderer 94  

Failed to load resource: the server responded with a status of 500 (Internal Server Error) :54776/api/printingQ…0?_=1609270009852:1

Uncaught (in promise) app.e40b4a8c3a5bf024…s?_=1608733777915:2 
g
req: v {_query: Array(0), method: "GET", url: "/api/printingQualityDefinitions/A250?_=1609270009852", header: {…}, _header: {…}, …}
xhr: XMLHttpRequest {readyState: 4, timeout: 0, withCredentials: false, upload: XMLHttpRequestUpload, onreadystatechange: ƒ, …}
text: "{"error":"XHR error"}"
statusText: "Internal Server Error"
statusCode: 500
status: 500
statusType: 5
info: false
ok: false
redirect: false
clientError: false
serverError: true
error: Error: cannot GET /api/printingQualityDefinitions/A250?_=1609270009852 (500) at g.toError (http://127.0.0.1:54776/481c6884/app.e40b4a8c3a5bf0241544.bundle.js?_=1608733777915:2:804007) at g.i._setStatusProperties (http://127.0.0.1:54776/481c6884/app.e40b4a8c3a5bf0241544.bundle.js?_=1608733777915:2:1975144) at new g (http://127.0.0.1:54776/481c6884/app.e40b4a8c3a5bf0241544.bundle.js?_=1608733777915:2:801982) at v.<anonymous> (http://127.0.0.1:54776/481c6884/app.e40b4a8c3a5bf0241544.bundle.js?_=1608733777915:2:802667) at v.r.emit (http://127.0.0.1:54776/481c6884/app.e40b4a8c3a5bf0241544.bundle.js?_=1608733777915:2:1041131) at XMLHttpRequest.t.onreadystatechange (http://127.0.0.1:54776/481c6884/app.e40b4a8c3a5bf0241544.bundle.js?_=1608733777915:2:806427)
created: false
accepted: false
noContent: false
badRequest: false
unauthorized: false
notAcceptable: false
forbidden: false
notFound: false
unprocessableEntity: false
headers: {connection: "keep-alive", content-length: "21", content-type: "application/json; charset=utf-8", date: "Tue, 29 Dec 2020 19:26:49 GMT", etag: "W/"15-L5Vt6IHsZy6lzu2EnC6jCR7LG6Y"", …}
header: {connection: "keep-alive", content-length: "21", content-type: "application/json; charset=utf-8", date: "Tue, 29 Dec 2020 19:26:49 GMT", etag: "W/"15-L5Vt6IHsZy6lzu2EnC6jCR7LG6Y"", …}
type: "application/json"
charset: "utf-8"
links: {}
body: {error: "XHR error"}
__proto__: Object

Failed to load resource: the server responded with a status of 500 (Internal Server Error) :54776/api/printingQ…0?_=1609270009952:1 

Uncaught (in promise) app.e40b4a8c3a5bf024…s?_=1608733777915:2 
g
req: v {_query: Array(0), method: "GET", url: "/api/printingQualityDefinitions/A250?_=1609270009952", header: {…}, _header: {…}, …}
xhr: XMLHttpRequest {readyState: 4, timeout: 0, withCredentials: false, upload: XMLHttpRequestUpload, onreadystatechange: ƒ, …}
text: "{"error":"XHR error"}"
statusText: "Internal Server Error"
statusCode: 500
status: 500
statusType: 5
info: false
ok: false
redirect: false
clientError: false
serverError: true
error: Error: cannot GET /api/printingQualityDefinitions/A250?_=1609270009952 (500) at g.toError (http://127.0.0.1:54776/481c6884/app.e40b4a8c3a5bf0241544.bundle.js?_=1608733777915:2:804007) at g.i._setStatusProperties (http://127.0.0.1:54776/481c6884/app.e40b4a8c3a5bf0241544.bundle.js?_=1608733777915:2:1975144) at new g (http://127.0.0.1:54776/481c6884/app.e40b4a8c3a5bf0241544.bundle.js?_=1608733777915:2:801982) at v.<anonymous> (http://127.0.0.1:54776/481c6884/app.e40b4a8c3a5bf0241544.bundle.js?_=1608733777915:2:802667) at v.r.emit (http://127.0.0.1:54776/481c6884/app.e40b4a8c3a5bf0241544.bundle.js?_=1608733777915:2:1041131) at XMLHttpRequest.t.onreadystatechange (http://127.0.0.1:54776/481c6884/app.e40b4a8c3a5bf0241544.bundle.js?_=1608733777915:2:806427)
created: false
accepted: false
noContent: false
badRequest: false
unauthorized: false
notAcceptable: false
forbidden: false
notFound: false
unprocessableEntity: false
headers: {connection: "keep-alive", content-length: "21", content-type: "application/json; charset=utf-8", date: "Tue, 29 Dec 2020 19:26:49 GMT", etag: "W/"15-L5Vt6IHsZy6lzu2EnC6jCR7LG6Y"", …}
header: {connection: "keep-alive", content-length: "21", content-type: "application/json; charset=utf-8", date: "Tue, 29 Dec 2020 19:26:49 GMT", etag: "W/"15-L5Vt6IHsZy6lzu2EnC6jCR7LG6Y"", …}
type: "application/json"
charset: "utf-8"
links: {}
body: {error: "XHR error"}
__proto__: Object
```

jane-rose, 2021-01-02:
Remove 'luban' config directory and restart the 'luban' will solve the problem.
On macOs:
`cd ~/Library/Application\ Support/snapmaker-luban`
`rm -rf Config/`
On Windows, just replace the config directory.

chrisywong, 2021-01-02:
I'm on Windows and not compiling from source. Is the config directory `C:\Program Files\Snapmaker Luban\resources\app\resources\CuraEngine\Config`? I deleted it and it still doesn't work. Replacing it with the same directory downloaded from this Github repo didn't work either.

chrisywong, 2021-01-02:
Never mind, deleted the folder `C:\Users\[Your Windows User Profile]\AppData\Roaming\snapmaker-luban` and then it worked again. Thanks for the solution! Though unfortunately I lost all my custom profiles. Going to see if I can salvage them from the deleted folder.

closed 2021-01-02 by chrisywong